raspiblitz/raspiblitz #2004: Help, can't run node

kotopuz, 2021-02-12:
I  have RPI 4 2GB. 32GB SSD and external HDD for blockchain.
Tried to burn image and it worked, but I was not able to use realVNC with desktop. Tried everything I can, I can't get it running.
Then I've followed instruction to manually burn image by running 
wget https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.6/build_sdcard.sh && sudo bash build_sdcard.sh
Then I've tried again from scratch by SUDO wget https://raw.githubusercontent.com/rootzoll/raspiblitz/v1.6/build_sdcard.sh && sudo bash build_sdcard.sh

After reboot I got message
-bash: home/admin_commands.sh: No such file in directory
-bash: ./00raspiblitz,sh: No such file in directory.

FIX IT! 1 day spent!
I follow all you brilliant instructions, but they are not working.
angry...

P.S. used raspi imager to burn raspi with desktop then tried with raspi full then tried lite .

NO RESULT

IF( NO REPLY) THEN PRINT ("never work with raspiblitz");
 else WHILE (1) {PRINT("THANK YOU SOOOO MUCH!");}

kotopuz, 2021-02-12:
One more thing. After running manua linstall I can't connect real VNC. I was able to do it prior install and even run update and full-upgrade. All was fine. After install resolution in raspi-cnofig  changed back to minimum setting! Why ?! then I've changes that back to min required by realVNC, but Iam still getting message "Cannon currently show the desktop". All suggestions from google not worked. Full upgrade did not work, reinstall of realvnc did not work.

Please take NEW latest image of raspian desktop and make new ISO with desktop included. You can do it in few hours and post it. Please ..

Ben10Brasileiro, 2021-02-12:
First, use SSH, not VNC, as the first is encrypted and the second is not.
Second, raspiblitz was made to run even headless, and a desktop enviroment consumes a lot of resources.
You can get all the funcitonality of blitz headless via SSH, HDMI, and a nice little LCD. But SSH does it all.
You don't need to build the image yourself, you can download the image ready:
Image https://raspiblitz.org/raspiblitz-v1.6.3-2021-01-13.img.gz
Instructions https://github.com/rootzoll/raspiblitz#downloading-the-software
 and check integrity and authority here https://github.com/rootzoll/raspiblitz/blob/v1.6/FAQ.md#how-to-verify-the-sd-card-image-after-download

nyxnor, 2021-04-18:
@kotopuz Please close this issue if it was solved. If you need any help, just ask.

nyxnor, 2021-05-17:
@rootzoll Please close this

closed 2021-05-18 by rootzoll