raspiblitz/raspiblitz #2008: Mempool Space backend crashes when bitcoind's mempool gets very large

wiz, 2021-02-14:
### Background

On a Raspberry Pi with 4GB or more of RAM, the bitcoin.conf setting for `maxmempool` is increase to 512MB.

### Issue

The increased size of the bitcoind mempool requires the mempool.space backend to utilize more RAM. Additionally, the mempool.space integration on RaspiBlitz reduces the setting for the max heap size of mempool backend's nodejs process from 4GB to 2GB. The combination of these two configuration settings causes the mempool.space backend to crash when the bitcoin mempool gets very full (over 400MB).

### Fixes

* Reduce the bitcoin.conf maxmempool setting back to 300MB as default
* Increase the mempool.space backend heap size back to 4GB as default
* Optimize the memory usage inside the mempool.space backend

wiz, 2021-02-14:
> Optimize the memory usage inside the mempool.space backend

Simon optimized the memory usage in the mempool backend in https://github.com/mempool/mempool/pull/342 and we will prepare a `v2.1.2` hotfix release containing that patch - from our testing it should now work fine with only 2G heap size

rootzoll, 2021-03-18:
faxes and updates merged ... set to final testing.

rootzoll, 2021-03-18:
OK first test looks good :) closing this issue ... final testing fro release on: #2076

closed 2021-03-18 by rootzoll

tortxoFFoxtrot, 2024-08-14:
It has been crashing for me and i suspect this is culprit 
[image]

tortxoFFoxtrot, 2024-08-14:
Well I put this in my compose.yml for the api

```
  environment:
      NODE_OPTIONS: --max_old_space_size=4096
```

And so far I am not seeing the messages anymore. Will update if bad things happen.

tortxoFFoxtrot, 2024-08-15:
Came to say that after applying some fixes it has never ran better.

I increased the node mem heap to 4096, I am using mariadb:latest as the db docker image and I am forcing the db collation to utf8 and it is running like a dream, has never ran better, this is on a 4 core J3160 celleron with 8GB of DDR3 ram.

Below docker compose, put inn your own IP/Ports and user/pass for your setup.

```
services:
  web:
    environment:
      FRONTEND_HTTP_PORT: "8080"
      BACKEND_MAINNET_HTTP_HOST: api
    image: mempool/frontend:latest
    user: 1000:1000
    restart: unless-stopped
    stop_grace_period: 1m
    command: ./wait-for db:3306 --timeout=720 -- nginx -g 'daemon off;'
    ports:
      - 127.0.0.1:8080:8080 # I am using nginx to expose this more broadly
  api:
    environment:
      #MEMPOOL_BACKEND: none
      MEMPOOL_BACKEND: electrum
      ELECTRUM_HOST: 10.x.x.x
      ELECTRUM_PORT: "xxxx"
      ELECTRUM_TLS_ENABLED: "true"
      CORE_RPC_HOST: 10.x.x.x
      CORE_RPC_PORT: "xxxx"
      CORE_RPC_USERNAME: xxxx
      CORE_RPC_PASSWORD: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
      DATABASE_ENABLED: "true"
      DATABASE_HOST: db
      DATABASE_DATABASE: mempool
      DATABASE_USERNAME: mempool
      DATABASE_PASSWORD: mempool
      STATISTICS_ENABLED: "true"
      NODE_OPTIONS: --max_old_space_size=4096
    image: mempool/backend:latest
    user: 1000:1000
    restart: unless-stopped
    stop_grace_period: 1m
    command: ./wait-for-it.sh db:3306 --timeout=720 --strict -- ./start.sh
    volumes:
      - ./data:/backend/cache
  db:
    command: --character-set-server=utf8mb4 --collation-server=utf8mb4_bin
    environment:
      MYSQL_DATABASE: mempool
      MYSQL_USER: mempool
      MYSQL_PASSWORD: mempool
      MYSQL_ROOT_PASSWORD: makeyourownrootpassword
    image: mariadb:latest
    user: 1000:1000
    restart: unless-stopped
    stop_grace_period: 1m
    volumes:
      - ./mysql/data:/var/lib/mysql
networks: {}
```

rootzoll, 2024-08-15:
@tortxoFFoxtrot  you are running mempool in a docker compose setup on a RaspiBlitz?

tortxoFFoxtrot, 2024-08-16:
> @tortxoFFoxtrot you are running mempool in a docker compose setup on a RaspiBlitz?

Yes do you like it? Use Ubuntu as your base OS, install docker, use Dockge to manage

rootzoll, 2024-08-16:
@tortxoFFoxtrot sure if that works for you - but thats then not part of the RaspiBlitz setup. These GitHub issues discuss how to fix/improve mempool on a RaspiBlitz that is not running mempool in a docker env.

PR #2007 (2021-03-18): Always set maxmempool to 300MB in bitcoin.conf
When the mempool fills up to around 500MB, this causes the mempool.space backend to crash on RPI due to OOM. Additionally, raising this above 300MB on a local RPI node arguably isn't a good idea anyway, because:

* On a Raspberry Pi running RaspiBlitz, it takes over 30 minutes for Bitcoin Core to load 478MB of transactions from disk in mempool.dat into RAM, and during this time the RPI's CPU utilization is maxed out
* If you have a 512MB mempool and all your peers have 300MB, you'll broadcast transactions that fit in your mempool but are rejected  [... cut]