raspiblitz/raspiblitz #2009: Failed to start after initial setup

DyrMooning, 2021-02-14:
After running the setup and getting through the whole process, after reboot the raspberry pi 4 wouldn't start the bitcoin services. 

Here is what it output. It said that it **Failed to start A high performance web server and a reverse proxy server.** (this was highlighted in red text in terminal).

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.6.3
chainnetwork: bitcoin / main
 16:36:45 up 4 min,  2 users,  load average: 1.93, 1.45, 0.67

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: inactive (dead)

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Sun 2021-02-14 16:18:35 GMT, end at Sun 2021-02-14 16:36:45 GMT. --
-- No entries --
1613320573 ERROR: Error found in Logs
1613320594 ERROR: Error found in Logs

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: inactive (dead)

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Sun 2021-02-14 16:18:35 GMT, end at Sun 2021-02-14 16:36:45 GMT. --
-- No entries --

*** LAST 30 LND INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log' for reading: No such file or directory

*** LAST NGINX LOGS ***
sudo journalctl -u nginx -b --no-pager -n20
-- Logs begin at Sun 2021-02-14 16:18:35 GMT, end at Sun 2021-02-14 16:36:45 GMT. --
Feb 14 16:33:25 raspberrypi systemd[1]: Starting A high performance web server and a reverse proxy server...
Feb 14 16:33:26 raspberrypi nginx[1125]: nginx: [emerg] BIO_new_file("/mnt/hdd/app-data/nginx/tls.cert") failed (SSL: error:02001002:system library:fopen:No such file or directory:fopen('/mnt/hdd/app-data/nginx/tls.cert','r') error:2006D080:BIO routines:BIO_new_file:no such file)
Feb 14 16:33:26 raspberrypi nginx[1125]: nginx: configuration file /etc/nginx/nginx.conf test failed
Feb 14 16:33:26 raspberrypi systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
Feb 14 16:33:26 raspberrypi systemd[1]: nginx.service: Failed with result 'exit-code'.
Feb 14 16:33:26 raspberrypi systemd[1]: **Failed to start A high performance web server and a reverse proxy server.**
--> CHECK CONFIG: sudo nginx -t
nginx: [emerg] BIO_new_file("/mnt/hdd/app-data/nginx/tls.cert") failed (SSL: error:02001002:system library:fopen:No such file or directory:fopen('/mnt/hdd/app-data/nginx/tls.cert','r') error:2006D080:BIO routines:BIO_new_file:no such file)
nginx: configuration file /etc/nginx/nginx.conf test failed

- TOUCHSCREEN is OFF by config
- Loop is OFF by config
- RTL is OFF by config
- Electrum Rust Server is OFF by config
- BTCPayServer is OFF by config
- LNbits is OFF by config
- Thunderhub is OFF by config
- SPECTER is OFF by config
- SPHINX is OFF by config

*** MOUNTED DRIVES ***
Filesystem     Type     1K-blocks    Used Available Use% Mounted on
/dev/root      ext4      30358348 5774632  23300408  20% /
devtmpfs       devtmpfs   4045052       0   4045052   0% /dev
tmpfs          tmpfs      4078844       0   4078844   0% /dev/shm
tmpfs          tmpfs      4078844    8672   4070172   1% /run
tmpfs          tmpfs         5120       4      5116   1% /run/lock
tmpfs          tmpfs      4078844       0   4078844   0% /sys/fs/cgroup
tmpfs          tmpfs        32768       0     32768   0% /var/cache/raspiblitz
/dev/mmcblk0p1 vfat        258095   53236    204859  21% /boot
tmpfs          tmpfs       815768       0    815768   0% /run/user/1000
tmpfs          tmpfs       815768       0    815768   0% /run/user/1001

*** WIFI Info ***
activated=0
connected=0
localip=''

*** HARDWARE TEST RESULTS ***
raspberryPi=4
powerFAIL=0
powerWARN=0
powerMIN='840000 microVolt'
tempFAIL=0
tempWARN=0
tempMAX='5500 centiGrad'

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.1.56'
tempCelsius='49.6'
uptime=296
upsStatus='OFF'
startcountBlockchain=0
bitcoinActive=0
bitcoinErrorShort='Error found in Logs'
bitcoinErrorFull='0'
# OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain.log
startcountLightning=0
lndActive=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=0

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

openoms, 2021-02-14:
Are you using the recommended parts? An incompatible USB disk adapter can be a source of all kind of issues.

Would recommend to start with a fresh SDcard image again.

DyrMooning, 2021-02-15:
Yes I am. It turned out that it started to work after a few restarts. It has been running stable since it was able to get past the startup issue.

closed 2021-02-21 by rootzoll