raspiblitz/raspiblitz #2010: Ran 'update RaspiBlitz', broke my node

woeisme, 2021-02-15:
I was attempting to fix a ThunderHub problem by updating it from the ncurses menu, I ran 'Patch RaspiBlitz' after that and before it shutdown to reboot I got the error messages 

>sudo: /home/admin/config.scripts/blitz.datadrive.sh: command not found /home/admin/XXshutdown.sh: line 56: [: -eq: unary operator expected

After the reboot the unit was stuck in the ncurses menu with: 
(there were also some terminal commands that flashed by too quickly to see, just something mentioning 00raspiblitz.sh I think)

>qqqqqqqqqqq Node is Syncing qqqqqqqqqqqqqqk
                                                      x  Blockchain Progress :  restarts          x
                                                      x  Lightning Progress  : waiting            x
                                                      x  Login with SSH for more details:         x
                                                      x  Use CTRL+c to EXIT to Terminal           x
                                                      mqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqj

What are my next steps to get it back to a bootable state and hopefully retain my configuration settings?
How long do I have before my active channels start closing or I lose funds? I've been backing them up to an attached pendrive.

woeisme, 2021-02-15:
Some additional terminal output 

>sudo: /home/admin/config.scripts/blitz.statusscan.sh: command not found
sudo: /home/admin/config.scripts/internet.sh: command not found
sudo: /home/admin/config.scripts/lnd.check.sh: command not found
/home/admin/80scanLND.sh: line 33: [: -eq: unary operator expected
/home/admin/80scanLND.sh: line 115: [: -eq: unary operator expected
/home/admin/80scanLND.sh: line 177: [: -gt: unary operator expected
^Cadmin@bilbo:~ $ sudo: /home/admin/config.scripts/network.monitor.sh: command not found

openoms, 2021-02-15:
Reflashing the SDcard is the way forward. The patch menu is for development, not a regular update method.

woeisme, 2021-02-15:
> 
> 
> Reflashing the SDcard is the way forward. The patch menu is for development, not a regular update method.

Can  I reflash the SD card using the upgrade version instructions will that retain my configuration / channels?

openoms, 2021-02-15:
> > Reflashing the SDcard is the way forward. The patch menu is for development, not a regular update method.
> 
> Can I reflash the SD card using the RaspiBlitz Release Update instructions and will that retain my configuration / channels?

yes, same as with all releases and as described in the FAQ.

woeisme, 2021-02-17:
Thanks @openoms as usual for your help. Flashed up to 1.6.3 and was suprised to see all my software installs still there (I didn't realize those lived on the HDD). 

While channels and services seem to be kept, attempting to access RTL / ThunderHub via Tor or clearnet fail.

openoms, 2021-02-17:
the installations are only restored from the saved settings on the disk in 
`/mnt/hdd/app-data` and `/mnt/hdd/raspiblitz.conf`

Your Nginx might be down. Try a restart and check what 
`sudo nginx -t`
and 
`sudo systemctl status nginx`
show?

woeisme, 2021-02-17:
Thats really strange, I went to work on this today and ssh'd in only to notice that ncurses over ssh was prompting me for the credential reset. Reset credentials and now everything seems accessible. Definitely took some random delay to come up however, I wonder why.

woeisme, 2021-02-18:
now some sites are loading normally (BTC-Explorer) while others are failing (ThunderHub and RTL)

Ran your nginx commands and process and config look fine.

openoms, 2021-02-18:
To debug individual services:
```
#How to #debug individual #services
#check the status:
sudo systemctl status SERVICE
#logs (CTRL +C to exit):
sudo journalctl -n100 -fu SERVICE 
#restart:
sudo systemctl restart SERVICE
```

service names:
btc-rpc-explorer
thunderhub
RTL

If you keep getting weird errors it might be a failing SDcard. Changing it might help and you can reflash at any point to rerun the recovery process.

woeisme, 2021-02-19:
Checked a day later and services are all active and functioning, maybe the node just needed some time to settle perhaps. 
Channels are all accounted for now as well.
Thanks again for all your help @openoms and all that you contribute. I'll close this issue now. Really pleased at how easy recovering the node went.

woeisme, 2021-02-19:
remember kiddos **The patch menu is for development, not a regular update method.!**

closed 2021-02-19 by woeisme