raspiblitz/raspiblitz #2011: Joinmarket 8.1: New wallet creates error when working from Joinmarket Menu

davterra, 2021-02-17:
I created a new wallet. Then, using the Joinmarket Menu if I click on "INFO"  and select the new wallet or try to run MAKER Yield generator (after moving funds to the new wallet) I get the error shown in the attachment.

If I exit to JM CLI and run: 
python wallet-tool.py <new_wallet_name>
or 
python yg-privacyenhanced.py <new_wallet_name>

everything works fine.
[image]

openoms, 2021-02-17:
You cannot just select the new wallet on the dialog screen, need to typo it's full filename like
`wallet.jmdat`

The message above shows that the wallet filename was not given.

davterra, 2021-02-17:
Ok. I'll try that. It appeared as if I could tab the highlighted area to where the new wallet name was shown and select it and go from there.

davterra, 2021-02-18:
I was AFK all day, but yes, typing in the wallet name in the box was the difference. I will close this, but I do think the fact that the highlighted area can be tabbed to the individual wallets in the list makes it appear as if they can be selected without typing the wallet name.

closed 2021-02-18 by davterra