raspiblitz/raspiblitz #2012: Termbin is not printing the all the debug logs

Ben10Brasileiro, 2021-02-18:
This is because there is a time out, termbin just run debug again with a time out, doesnt get the previous outpu.
The best solution is to save a actual log file of the debug, when first running the debug command, so the user doesnt need to run twice. Also, this file will be replaced everytime, as it will be saved with the same name, so dont need to worry about increasing space.
command > file will also need to run again the command.

The `debug` command should include printing the ouputs to a file when initially been run
`debug > debug.txt` works, but I would like to trigger this command to automatically print everytime I ran debug, the new debug file, with **only typing debug**.
I didnt find an easier way than the described above, I just want to shorten this to only need to type `debug` and then `/home/admin/debug.txt | nc termbin.com 9999`

rootzoll, 2021-02-21:
Thanks :) Will take a look for the next release.

nyxnor, 2021-04-08:
Well, first decent pull request.

closed 2021-04-11 by nyxnor