raspiblitz/raspiblitz #2013: add /etc/tor/torrc -> /mnt/hdd/app-data  for recovery / upgrade process.

woeisme, 2021-02-19:
Seems like a good candidate as all custom hidden servicies (Zap & Zeuz wallet HS configurations) are located there. 

Additional (BTC-Explorer, RTL, Thunderhub) Services are already being re-added to /etc/tor/torrc when recovering / upgrading nodes just unsure what process writes to that file.

openoms, 2021-02-19:
the existing hidden service privatekeys are stored on the disk  in `/mnt/hdd/tor/` if they are present already the keys are not recreated and the .onion-s will stay the same after an update /recovery. 
They are not stored in the app-data directory so would not be a part of a MIGRATION file. This can be improved. Thanks for the suggestion.

nyxnor, 2021-04-18:
Also, I added bridges file on my pull requests, this should be saved also. It is located in `/etc/tor/bridges`. It is just a backup of bridges.

woeisme, 2021-05-22:
I just upgraded from 1.6.3 ->1.7 and needed to reset keys and re-import into Zeus to get working again. LND service line still remained however. 
 
Additionally I wasn't able to run lndconnect after upgrade and had to run 

> $ config.scripts/bonus.lndconnect.sh zeus-android tor

which is only really documented if you know about openoms separate tutorials.

rootzoll, 2021-06-14:
Will recheck the connectivity for v.1.7.1 release.

rootzoll, 2021-09-16:
Because this wil be part of the bigger Tor updates that we needed to push to v1.8 .. the v1.7.1 might still have those issues.

woeisme, 2021-11-12:
still losing user added hidden services in 1.7.0 upgrade.

nyxnor, 2021-11-18:
It is on the work

rootzoll, 2021-12-11:
OK I merged the changes of @nyxnor - that the data now survives updates/recoveries is to be final tested.

rootzoll, 2022-02-17:
services config is stored under in `/mnt/hdd/app-data/tor/torrc.d/services` - while /mnt/hdd/tor has the keys

closed 2022-02-17 by rootzoll