raspiblitz/raspiblitz #2015: Cant connect to RPC server from local network

frangb, 2021-02-21:
Hi all

Everything seems to be working fine in my node except for the connection via RPC from local network

I have changed the following two lines in .bitcoin/bitcoin.conf file (my raspiblitz local IP is 192.168.1.12)
```
rpcallowip=192.168.1.1/255.255.255.0
main.rpcbind=192.168.1.12:8332
```
after that I have added  the following rule to firewall
`sudo ufw allow 8332
`

and I have restarted the service with
`sudo systemctl restart bitcoind`

But still can't connect from other computers in the same LAN. When I try to connect I get the following response
```
telnet 192.168.1.12 8332
Trying 192.168.1.12...
telnet: connect to address 192.168.1.12: Connection refused
telnet: Unable to connect to remote host
```

I can connect to other ports in raspiblitz (for example 3000), but not to 8332

Any idea what's wrong with my setup?

rootzoll, 2021-02-21:
Basically the setup looks ok. But you can try the following:

- try with removed `rpcallowip`
- check logs if you can see any strange `sudo tail -n 100 /mnt/hdd/bitcoin/debug.log`

frangb, 2021-02-21:
Thanks for the reply. I have tried removing `rpcallowip` but nothing has changed

Here is the log, I can't see any error there

```
2021-02-21T12:04:24Z Default data directory /home/bitcoin/.bitcoin
2021-02-21T12:04:24Z Using data directory /mnt/hdd/bitcoin
2021-02-21T12:04:24Z Config file: /home/bitcoin/.bitcoin/bitcoin.conf
2021-02-21T12:04:24Z Config file arg: daemon="1"
2021-02-21T12:04:24Z Config file arg: datadir="/mnt/hdd/bitcoin"
2021-02-21T12:04:24Z Config file arg: dbcache="512"
2021-02-21T12:04:24Z Config file arg: disablewallet="1"
2021-02-21T12:04:24Z Config file arg: dns="0"
2021-02-21T12:04:24Z Config file arg: dnsseed="0"
2021-02-21T12:04:24Z Config file arg: maxconnections="40"
2021-02-21T12:04:24Z Config file arg: maxmempool="300"
2021-02-21T12:04:24Z Config file arg: maxorphantx="10"
2021-02-21T12:04:24Z Config file arg: maxuploadtarget="5000"
2021-02-21T12:04:24Z Config file arg: onlynet="onion"
2021-02-21T12:04:24Z Config file arg: peerbloomfilters="1"
2021-02-21T12:04:24Z Config file arg: proxy="127.0.0.1:9050"
2021-02-21T12:04:24Z Config file arg: rpcallowip="127.0.0.1"
2021-02-21T12:04:24Z Config file arg: rpcpassword=****
2021-02-21T12:04:24Z Config file arg: rpcuser=****
2021-02-21T12:04:24Z Config file arg: server="1"
2021-02-21T12:04:24Z Config file arg: testnet="0"
2021-02-21T12:04:24Z Config file arg: torpassword=****
2021-02-21T12:04:24Z Config file arg: txindex="1"
2021-02-21T12:04:24Z Config file arg: upnp="0"
2021-02-21T12:04:24Z Config file arg: zmqpubrawblock="tcp://127.0.0.1:28332"
2021-02-21T12:04:24Z Config file arg: zmqpubrawtx="tcp://127.0.0.1:28333"
2021-02-21T12:04:24Z Config file arg: [main] addnode="fno4aakpl6sg6y47.onion"
2021-02-21T12:04:24Z Config file arg: [main] addnode="toguvy5upyuctudx.onion"
2021-02-21T12:04:24Z Config file arg: [main] addnode="ndndword5lpb7eex.onion"
2021-02-21T12:04:24Z Config file arg: [main] addnode="6m2iqgnqjxh7ulyk.onion"
2021-02-21T12:04:24Z Config file arg: [main] addnode="5tuxetn7tar3q5kp.onion"
2021-02-21T12:04:24Z Config file arg: [main] addnode="juo4oneckybinerq.onion"
2021-02-21T12:04:24Z Config file arg: [main] bind="127.0.0.1"
2021-02-21T12:04:24Z Config file arg: [main] rpcbind=****
2021-02-21T12:04:24Z Config file arg: [main] rpcport="8332"
2021-02-21T12:04:24Z Config file arg: [test] bind="127.0.0.1"
2021-02-21T12:04:24Z Config file arg: [test] rpcbind=****
2021-02-21T12:04:24Z Config file arg: [test] rpcport="18332"
2021-02-21T12:04:24Z Command-line arg: conf="/home/bitcoin/.bitcoin/bitcoin.conf"
2021-02-21T12:04:24Z Command-line arg: daemon=""
2021-02-21T12:04:24Z Command-line arg: pid="/home/bitcoin/.bitcoin/bitcoind.pid"
2021-02-21T12:04:24Z Using at most 40 automatic connections (1024 file descriptors available)
2021-02-21T12:04:24Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2021-02-21T12:04:25Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2021-02-21T12:04:25Z Script verification uses 3 additional threads
2021-02-21T12:04:25Z scheduler thread start
2021-02-21T12:04:25Z Wallet disabled!
2021-02-21T12:04:25Z HTTP: creating work queue of depth 16
2021-02-21T12:04:25Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2021-02-21T12:04:25Z HTTP: starting 4 worker threads
2021-02-21T12:04:25Z init message: Loading banlist...
2021-02-21T12:04:25Z Using /16 prefix for IP bucketing
2021-02-21T12:04:25Z Cache configuration:
2021-02-21T12:04:25Z * Using 2.0 MiB for block index database
2021-02-21T12:04:25Z * Using 63.8 MiB for transaction index database
2021-02-21T12:04:25Z * Using 8.0 MiB for chain state database
2021-02-21T12:04:25Z * Using 438.2 MiB for in-memory UTXO set (plus up to 286.1 MiB of unused mempool space)
2021-02-21T12:04:25Z init message: Loading block index...
2021-02-21T12:04:25Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2021-02-21T12:04:25Z Opened LevelDB successfully
2021-02-21T12:04:25Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000
2021-02-21T12:04:38Z LoadBlockIndexDB: last block file = 2461
2021-02-21T12:04:38Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=56, size=89762388, heights=671501...671556, time=2021-02-21...2021-02-21)
2021-02-21T12:04:38Z Checking all blk files are present...
2021-02-21T12:04:38Z Opening LevelDB in /mnt/hdd/bitcoin/chainstate
[... 11 lines cut ...]
2021-02-21T12:04:48Z Opening LevelDB in /mnt/hdd/bitcoin/indexes/txindex
2021-02-21T12:04:49Z Opened LevelDB successfully
2021-02-21T12:04:49Z Using obfuscation key for /mnt/hdd/bitcoin/indexes/txindex: 0000000000000000
2021-02-21T12:04:49Z txindex thread start
2021-02-21T12:04:49Z txindex is enabled at height 671556
2021-02-21T12:04:49Z txindex thread exit
2021-02-21T12:04:49Z block tree size = 671575
2021-02-21T12:04:49Z nBestHeight = 671556
2021-02-21T12:04:49Z torcontrol thread start
2021-02-21T12:04:49Z Bound to 127.0.0.1:8333
2021-02-21T12:04:49Z init message: Loading P2P addresses...
2021-02-21T12:04:49Z Leaving InitialBlockDownload (latching to false)
2021-02-21T12:04:50Z tor: Got service ID 7t2yx4tbnzs4uzcl, advertising service 7t2yx4tbnzs4uzcl.onion:8333
2021-02-21T12:04:50Z AddLocal(7t2yx4tbnzs4uzcl.onion:8333,4)
2021-02-21T12:04:50Z Loaded 64241 addresses from peers.dat  692ms
2021-02-21T12:04:50Z init message: Starting network threads...
2021-02-21T12:04:50Z DNS seeding disabled
2021-02-21T12:04:50Z init message: Done loading
2021-02-21T12:04:50Z net thread start
2021-02-21T12:04:50Z addcon thread start
2021-02-21T12:04:50Z opencon thread start
2021-02-21T12:04:50Z msghand thread start
2021-02-21T12:05:01Z New outbound peer connected: version: 70015, blocks=671556, peer=0 (full-relay)
2021-02-21T12:05:10Z New outbound peer connected: version: 70015, blocks=671556, peer=1 (full-relay)
```

rootzoll, 2021-02-21:
In the beginning of the logs when the config is loaded I can only see one rpcbind for mainet --> '2021-02-21T12:04:24Z Config file arg: [main] rpcbind=****' is that the line that contains you `main.rpcbind=192.168.1.12:8332`?

Also you can have multiple `main.rpcbind` entries ... I would always also leave `main.rpcbind=127.0.0.1:8332` in there so that all raspiblitz services are able to reach the RPC service.

rootzoll, 2021-02-21:
Ah I please also check the config file here "/mnt/hdd/bitcoin/bitcoin.conf" ... the files in `.bitcoin/bitcoin.conf` is just for the bitcoin-cli of local users.

frangb, 2021-02-21:
You are a genius man! just solved the problem by editing the file in `/mnt/hdd/bitcoin/bitcoin.conf` instead of the one in the home directory!

Many thanks!

closed 2021-02-21 by frangb