raspiblitz/raspiblitz #2020: custom-installs.sh Script to Create Hidden Service Address for SSHD

davterra, 2021-02-23:
I've been using (and LOVING) my Raspiblitz for less than a week, but I was surprised to see that there's no apparent Tor address for SSHD. This would allow SSH access from outside your home network without opening/forwarding a WAN port on your router.

After watching Rootzol's great YouTube video on "How to Contribute", I made a simple script to add the SSHD Hidden Address, and tested it. It seems to me to be a feature that is "ON" by default, so no Menu toggle necessary. 

My coding skills are "lower-intermediate", so when I went to add the SSHD Hidden Service hostname to the INFO page, I wasn't confident that I could do it. If someone else wants to take this further and create a PR, go ahead...but my preference would be to learn how to do it myself. 
I know how to create a Github branch and a PR, I just don't yet have a grasp of the RaspiBlitz file structure to go much further on my own.

Here is the script:
########################################################
# Script to create HiddenService for SSHD
sed -i '$ a\ ' /etc/tor/torrc
sed -i '$ a\# Hidden Service for SSH via Tor' /etc/tor/torrc
sed -i '$ a\HiddenServiceDir /mnt/hdd/tor/sshd' /etc/tor/torrc
sed -i '$ a\HiddenServiceVersion 3' /etc/tor/torrc
sed -i '$ a\HiddenServicePort 22 127.0.0.1:22' /etc/tor/torrc
sudo systemctl restart tor
########################################################

Ben10Brasileiro, 2021-02-23:
Are you refering to SSH?

https://twitter.com/openoms/status/1361449493015777285

There is this project Shellinabox:
```
sudo apt install shellinabox
config.scripts/internet.hiddenservice.sh shellinabox 443 4200 
```
* Open the https://.onion in the Tor Browser

Disclaimer: it is what you would expect of onion routing, slow but secure. Dont let anyone see this address.
To access from outside home you have to use a service that provide this connection. As VPN Zero Tier is recommended and implemented on the menu, which you can set to route directly from around the globe if you have a machine in each location.
https://www.zerotier.com/

davterra, 2021-02-23:
> Are you refering to SSH?
> 
> https://twitter.com/openoms/status/1361449493015777285
> 
> There is this project Shellinabox:
> 
> ```
> sudo apt install shellinabox
> config.scripts/internet.hiddenservice.sh shellinabox 443 4200 
> ```
> 
> * Open the https://.onion in the Tor Browser
> 
> Disclaimer: it is what you would expect of onion routing, slow but secure. Dont let anyone see this address.
> To access from outside home you have to use a service that provide this connection. As VPN Zero Tier is recommended and implemented on the menu, which you can set to route to your machines if you have a machine in each location.
> https://www.zerotier.com/
> 
> I didnt try ZeroTier yet, still reading the documentation.

Yes, I was referring to the SSHdaemon that is given a hidden address.

The Shellinabox looks interesting. I'd like to hear from anyone who has tried it.

Ben10Brasileiro, 2021-02-23:
I tried...
It is nice to have this feature, quite dangerous if you cant change the onion address per option. But if you would like to search how to change the onion address if the users choose to, I would like to try it out.
Nothing bad to say about it, just not been developed anymore.

davterra, 2021-02-23:
Yeah, I think my needs are met by just the script I wrote. I dont like using privacy software that is no longer being maintained.

rootzoll, 2021-02-25:
Hi @davterra thanks for sharing your script - that is the way :)

First of all there is a config-script of RaspiBlitz you can use to expose the ssh port 22 with tor thru a oneliner:
`/home/admin/config.scripts/internet.hiddenservice.sh` see here: https://github.com/rootzoll/raspiblitz/blob/v1.6/home.admin/config.scripts/internet.hiddenservice.sh

So it should be '/home.admin/config.scripts/internet.hiddenservice.sh ssh 22 22' (cant test I am on the road)

The script will let you know the created onion-address and the config it will surive an update or migration.

To make this default .. I am not completly convinced, because I think its good practice to just open such important services to the public if needed. But to make izt more easy - for example to add this options to the SETTINGS menu can make much sense.

rootzoll, 2021-02-25:
@Ben10Brasileiro ah OK you already pointed him to the config script :) thx

Ben10Brasileiro, 2021-02-25:
>To make this default .. I am not completly convinced, because I think its good practice to just open such important services to the public if needed. But to make izt more easy - for example to add this options to the SETTINGS menu can make much sense.

I agree with you, just as an option on menu but not default.

davterra, 2021-02-27:
Yeah, I'm fine with just an option on the menu.

nyxnor, 2021-05-17:
Useful issue about concerns https://github.com/rootzoll/raspiblitz/issues/2273
Check to persist /mnt/hdd/tor/ssh after update https://github.com/rootzoll/raspiblitz/issues/2306
Important note on the last issue: add client auth (+1 layers of security). This also can be used for the webUI, the key will be entered in the browser.

openoms, 2021-05-18:
The tor folder persists, only the torrc file doesn't.
Changing the SDcard needs physical access already so there is no point persisting the remote access (and a security concern also: https://github.com/rootzoll/raspiblitz/issues/2306#issuecomment-843460518)

nyxnor, 2021-05-18:
About security concern, answered on the above mentioned issue. About changing the sd card, correct.