raspiblitz/raspiblitz #2021: Create tar.gz-Backupfile

Dinglebarry9, 2021-02-24:
I have an older lnd node with a bunch of open channels that I would like to migrate to raspiblitz. Would it be better to create the tar.gz-Backupfile with the contents of the .lnd folder or to manually place it in the /home/admin/.lnd directory of the raspiblitz by hand?

rootzoll, 2021-02-24:
The "tar.gz-Backupfile" and use that as the migration file on fresh setup would be the most clean way. But its a bit hard to get right ... because on import also checks the checksum that is part of the name and the data structure inside needs to be correct. Best way would be:

- setup a fresh raspiblitz and then create a migration file (tar.gz-Backupfile) from it REPAIR > MIGRATION
- unpack that file on your laptop and replace the lnd files
- make it a tar.gz file again with the same name
- make a md5 checksum of that tar.gz file and replace the old md5checksum part in the tar.gz file name

This is the strcture how the final file needs to be named `  name="raspiblitz${blitzname}${datestamp}-${md5checksum}.tar.gz"`

Then reset the fresh RaspiBlitz with REPAIR > RESET-HDD (which keeps the blockchain), make a fesh sd card and this time choose "MIGRATION" during setup and upload your prepared file.

Dinglebarry9, 2021-02-24:
Awesome, I'll give it a go thanks.

closed 2021-02-24 by Dinglebarry9