raspiblitz/raspiblitz #2023: Waiting for internet

velgard1, 2021-02-25:
I have version 1.6.3 and it worked fine for a while, but since some days the internet connection is lost (I use the Raspiblitz with TOR).
First I thought this is an issue with the router, but all other devices run fine - I also can access the Raspiblitz via SSH (display also says that network connection is fine but no internet).
My plan was to access the main menu, but it doesn't work and I also found no command to restart or shutdown from the terminal...
So from what I found I would now unplug the network cable, wait a while, unplug the power and then try to restart with the cable connected - this seems to be save, right?

rootzoll, 2021-02-25:
Yes its safe to unplug & replug the network cable. Let me know if this fixes it. Also try a reboot .. you may need to CTRL+C aout of the dialog screens to get to the terminal and then use command `restart` for reboot.

velgard1, 2021-02-26:
Plug & unplug did not help - the cable & ports in general should be ok since SSH is working.
After restarting I have the same issue, it's not connected to peers acc. to the display.
I also restarted the router (fritzbox), but to no avail.
My guess is that I somehow have to connect to peers via CL manually (I use TOR). Is there an explanation how to do this somewhere?

Ben10Brasileiro, 2021-03-02:
@velgard1 Are you still having this issue?

velgard1, 2021-03-02:
Yes, still the same.
My main issue is that I can't access the menu bacause of it and I'm a complete newbie to Linux CLI...

Ben10Brasileiro, 2021-03-07:
@velgard1 

As you said you can ssh, but the signal is weak, here are 2 methods.

#### Connect ethernet cable
* This is the best option, less signal loss, more consistency. I dont have a cable from my router to my Pi, and the Wifi was constantly disconnecting, so I did the following (Dont judge me for this mess, it was the only option).
The Pi has weak wifi when connected to a large storage that demans more power than the microSD card, so this Wifi Extender gets the wifi signal, and connects to the Pi vi ethernet cable, on `ifconfig` it receives eth0 IP address. I also enabled wifi as the second method just in case.
[image]

#### Enabling WIFI:

If you have the microSD card still plugged on your computer, drop a file named wifi on the /boot folder with the lines below.

If you already are running the Raspiblitz do: sudo nano /etc/wpa_supplicant/wpa_supplicant.conf and add the lines below. Info: change the country US to your own country ISO Alpha-2 code. Search the ISO code here https://www.iso.org/obp/ui#search Watch this guide for more support: https://youtu.be/77BBQWg1n8w?t=398

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US
network={
        scan_ssid=1
        ssid="wifiName"
        psk="password"
}

velgard1, 2021-03-14:
It works again, but I have no idea why...
After my last post I managed to shutdown via CLI and only had time to work on the issue today - booted, connected and running fine without any issue.
Regarding the methods you mentioned I already had a direct LAN cable to my router from the Raspiblitz, only my normal PC is connected via WiFi, so I suppose it was no connection issue.
Thank you anyway, it's great to know how to access the main menu via CLI if this should be necessary some day!

closed 2021-03-14 by velgard1