raspiblitz/raspiblitz #2024: Lost all address labels after update

romfordpele, 2021-02-25:
Hi guys,

Hope I can explain this issue well and potentially point out how my cock up may have cost me all my address labels.

So had my node up and running for maybe 5 months. I use a hardware wallet and have backups made of this. I use specter wallet to manage the UTXOs but crucially I don't think it ever actually 'clicked' with me that I would need to backup my WALLET. I have never used my own node before so never had to worry about third party wallet software before now.

Raspiblitz update comes along and I update the blitz to the latest version. When I boot it up I see specter needs to be restored....makes sense. I restore the wallet from my hardware wallet backup file and when the wallet is synced I realise that my coins are there but all my labels have disappeared. They have reverted to address 1, address 2 etc.

It has occurred to me that if I had backed up the WALLET (i.e. the specter backup), and restored from that, then it would contain all the labels as they should be. 

I have the old SD card and tried to boot the blitz from that in the hope that I could recover the old specter with the labels, make a backup to restore from and I would be saved. Unfortunately the blitz would not boot from the old version. 

Now I'm stuck and asking if anyone has any other ideas to get the address labels back! I am wondering if the labels are held on the old SD card somewhere I can view them. Looking in this way may not recover them automatically but least I could see which addresses were allocated which labels and manually re-enter them?

Hope I'm making sense! Would be mega grateful if anyone had any ideas.

This project is amazing and I hope one day I can give back to this community in some small way. It took me a few months to actually get my raspiblitz up and validating my transactions, just being able to do this is a great feeling. I've not even started playing around with lightning yet. Top marks to everyone working their asses off to make it possible.

mmarsalko, 2021-03-01:
If you plug the SD into a linux computer, you should be able to browse the filesystem and find the file that contains your address labels. Look in /home/admin/.specter/wallets -- there should be a <something>_addr.csv in there. Open up the files in there and see which ones contain your address labels, then copy that file back over to your blitz.

romfordpele, 2021-03-06:
Thanks for the reply! I run on MacOS so not sure if this is possible without changing my OS to Linux?

mmarsalko, 2021-03-07:
MacOS should be able to read the SD just fine.

nyxnor, 2021-04-18:
@romfordpele  You should asking in the specter github. Can you close this? If you have any enlightinment later, feel free to open this again.

romfordpele, 2021-05-13:
I did ask the specter guys and they sent me here.

I have an update though...

Installed a Linux virtual machine on the mac and long story short managed to find two locations on the sd card - "boot", which I could see on the mac and looked exactly the same in terms of contents. And "rootfs" which I'd never seen on mac.

In rootfs there was the /home path described above. Trawled through there and couldn't find specter at all. However a bit more googling suggested hidden files and folders begin with a . so worked out how to reveal these. Found the .specter folder in the home/.bitcoin/ path! 

Unfortunately there was no .csv file at all. So I've failed to find the label and think I'm giving up on it now. On the positive side I've learnt how to use virtual machine and have a better idea of what linux can do and a greater respect for backups etc.

Will close this issue thanks for the assistance.

closed 2021-05-13 by romfordpele

openoms, 2021-05-13:
As with most services the data is stored on the disk in:
`/mnt/hdd/app-data/.specter`
https://github.com/rootzoll/raspiblitz/blob/be19c565cb5fb570abcd7cf6ad83cc5e4b840520/home.admin/config.scripts/bonus.cryptoadvance-specter.sh#L121

and symlinked to the service user ome directory on the SDcard:
https://github.com/rootzoll/raspiblitz/blob/be19c565cb5fb570abcd7cf6ad83cc5e4b840520/home.admin/config.scripts/bonus.cryptoadvance-specter.sh#L125