raspiblitz/raspiblitz #2025: Change Font Size On Raspiblitz

pkepler, 2021-02-26:
I have searched "issues" for changing font size and haven't found the fix for this.  How do I make this fill the entire screen?  Thanks in advance.
[image]

pkepler, 2021-02-26:
I have reopened this issue after some Telegram suggestions.  Those steer me to changing font size but I think this looks like the Raspiblitz software is sending out a higher resolution than this 4 inch screen is set to.  I think I need to lower the resolution on the pi.  I appreciate any help.  My CLI interface is weak but I am learning so if you help please be specific.  Thank you.

NilByte, 2021-02-27:
I would try `sudo raspi-config`

https://pimylifeup.com/raspberry-pi-screen-resolution/

pkepler, 2021-02-27:
I did try that.  This LCD is the 4 inch LCD  http://www.lcdwiki.com/4inch_HDMI_Display-C.  Shows a native resolution of 800 * 480.    I tried all of the options in the menu.  None of them filled the screen.  Most appeared to be incompatible as the screen just kept flashing blue screen.  I found this in another thread:

# Force use HDMI
hdmi_force_hotplug=1
# force a specific HDMI mode
hdmi_group=2
hdmi_mode=87

# custom display resolution (480x320 looks good , and 720x480 is not too bad too,3rd res is: 810x540 )
#         /----------------- Width: 480px ( or 720px /or 810px)
#         |   /------------- Height: 320px ( or 480px /or 540px)
#         |   |   /--------- Refresh: 60Hz
#         |   |   | /------- Aspect ratio: 6=15:9 (1=4:3, 2=14:9, 3=16:9, 4=5:4, 5=16:10)
#         |   |   | | /----- Margins: 0=disabled (1=enabled)
#         |   |   | | | /--- Interlace: 0=progressive (1=interlaced)
#         |   |   | | | | /- Reduced blanking: 0=normal (1=reduced)
#         |   |   | | | | |
hdmi_cvt 480 320 60 6 0 0 0

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
hdmi_drive=2

Even though this is for a 3.5 inch screen I think I will try it.

openoms, 2021-04-04:
@pkepler try this solution: https://github.com/rootzoll/raspiblitz/issues/244#issuecomment-476713706

nyxnor, 2021-04-18:
@pkepler Close this if solved pls.

pkepler, 2021-04-18:
Problem not solved but I don't care any longer.

closed 2021-04-18 by pkepler