raspiblitz/raspiblitz #2028: Scripts to automate an email send of your LND Channel.Backup file upon any change/update

davterra, 2021-02-28:
RaspiBlitz has an option to upload your CHannel.Backup to Dropbox, but this simple process will automatically email a fresh Channel.Backup to whatever email address you stipulate, any time the backup file is changed/updated.

Details in this gist:
https://gist.github.com/davterra/8ae5753a2b7b6d9a63d4872139a6a64d

davterra, 2021-03-01:
NOTE: I found a couple of typos in the gist, and fixed at 14:50 UTC. If anyone copied the files before that, please re-copy.

frennkie, 2021-03-01:
We already have the ability to setup email notifications.
So that part of of the gist isn't needed.

davterra, 2021-03-02:
Really? I'm new to this project but I only have seen the DropBox and USB backup options. Where can I find info about an existing email SCB/backup setup?

Ben10Brasileiro, 2021-03-07:
@davterra config.scripts/blitz.notify.sh

rootzoll, 2021-03-08:
@davterra is the existing notify script working for you? Can we close this issue?

davterra, 2021-03-08:
@rootzoll Sure, you can close it. I'm curious as to why this existing feature remains undocumented? I haven't tested it because my method is working fine for me. I have tried ssmtp in the past and ended up using mutt because I found it easier to configure, but it looks like you have that all scripted out. If it needs documentation, I would be willing to test it and add documentation to the README, if you want.

rootzoll, 2021-03-08:
Its not well documented because its still an experimental feature added by @frennkie .. adding some more documentation is much welcome :)

davterra, 2021-03-08:
OK. I'm on the road all of this week, so likely won't get started testing the existing script until next week, but happy to give it a try and write something up.

closed 2021-03-15 by davterra