raspiblitz/raspiblitz #2030: Adding liquid to btcpay & raspiblitz deployment

Xaspr, 2021-02-28:
Hi, 
Everything I wanted to use just worked right out of the box, which is great. 
Now I'm also looking into adding accepting Liquid BTC payments in btcpay. It seems to be pretty easy to do this with a docker deployment, but with the deployment in the Raspiblitz I can't figure this out yet. Could someone point me in the right direction?

rootzoll, 2021-03-08:
RaspiBlitz has no support to run addtional dockerized apps (yet). So if you want todo this with docker i would recommend checking back for a later version.

Xaspr, 2021-03-10:
Thanks, I'll definitely keep up to date on the future versions. 

For now I'll see how to get this to work with the manual setup of btcpay.  When I figure it out, I'll post it here for future reference.

nyxnor, 2021-05-17:
@Xaspr ping

Xaspr, 2021-05-19:
I haven't exactly applied this. Instead I ran the BTCPay Raspberry Pi 4 deployment on a separate Pi. 

A few links on this: 

How to add Liquid in BTCPay docker installation, so not directly applicable to raspliblitz: https://github.com/btcpayserver/btcpayserver/issues/1282

If you choose to install via manual deployment (so on a raspiblitz for example), you would need to install a liquid node manually on the Pi. 
https://help.blockstream.com/hc/en-us/articles/900002026026-How-do-I-set-up-a-Liquid-node-
And these are some of the settings that you would need: 
https://github.com/btcpayserver/btcpayserver-docker/blob/master/docker-compose-generator/docker-fragments/liquid.yml

I hope this is of any help.

cstenglein, 2023-06-07:
closed in favor of #2103