raspiblitz/raspiblitz #2031: Deprecate LNDmanage

rootzoll, 2021-03-01:
Some users report that LNDmanage is making problems on their nodes. Also the project is not very active anymore and most of the features are possible with other tools. 

May consider a soft removal for next release:
- remove from GUI
- deactivate on update (force turn off on update, even if it was on before  - so user needs to run command line install manually to reactivate)

keblek, 2021-03-05:
this might be a chance to add lndmon https://github.com/lightninglabs/lndmon

openoms, 2021-03-05:
LNDmanage is a python based rebalancing tool. Nothing to do with lndmon.

keblek, 2021-03-05:
I use bos for that so I wasn't familiar with this. My bad.

rootzoll, 2021-03-22:
removed it from GUI and update-provisiong for next release .. we can switch it back on if a new release is comming
changes made in branch https://github.com/rootzoll/raspiblitz/tree/2089-sphinx .. waiting to merge

closed 2021-04-17 by rootzoll

DocBrown101, 2021-05-13:
The new release is now available:
https://github.com/bitromortac/lndmanage/releases/tag/v0.12.0

rootzoll, 2021-05-18:
@DocBrown101 I opened an issue to update it for v1.7.1 see #2314