raspiblitz/raspiblitz #2032: Zap iOS officially unmaintained

Fittiboy, 2021-03-01:
Zap iOS status has officially changed to [unmaintained](https://github.com/LN-Zap/zap-iOS/commit/2a2f9630255213e308075107964d8b6bd1d911d1).  
It might be a good idea to reflect this somewhere in the menus, or remove the option to connect to Zap iOS from the menu entirely.

keblek, 2021-03-02:
thanks for the heads up - it was a great wallet, why did they stop?

Fittiboy, 2021-03-02:
Looking at [the commit](https://github.com/LN-Zap/zap-iOS/commit/2a2f9630255213e308075107964d8b6bd1d911d1):
```diff
- ## Maintainers
-
- - [Otto Suess (@ottosuess)](https://github.com/ottosuess)
-
```
Only Otto was left maintaining the iOS version. He has since moved on to [working on Strike](https://twitter.com/ottosuess/status/1362780778988523521).

keblek, 2021-03-02:
makes sense, I don't think there are any wallets that connect to and on iOS now
I guess there is bluewallet but that uses lnd hub and its not native

Fittiboy, 2021-03-02:
[Zeus](https://github.com/ZeusLN/zeus) works, I just switched!
In the raspiblitz menu, you have to select Zeus Android to connect to it (at least via Tor, not sure about clearnet).

keblek, 2021-03-02:
its not on iOS though

keblek, 2021-03-02:
yes it is, its on testflight

rootzoll, 2021-03-25:
OK I removed the Zap-iOS from the mobile Wallet options for now - its still in the code and could be tryied by command line but its not displayed in the SSH GUI for now.

Will push issue to v1.7.1 to take a look again of the state of the project then.

rootzoll, 2021-09-20:
Will push issue to v1.8.0 to take a look again of the state of the project then.

rootzoll, 2022-03-09:
Zap was reintroduced as an option into v1.7.2

closed 2022-03-09 by rootzoll