raspiblitz/raspiblitz #2036: [Enhancement] Create all the temporary files in RAM

openoms, 2021-03-04:
As discussed in https://github.com/openoms/joininbox/issues/16#issuecomment-790634039 and done in https://github.com/openoms/joininbox/commit/8449240919c0ce4cbb3bca9dffe1d3f546b8b5ad

Use:
`wallet=$(mktemp -p /dev/shm/)`
instead of
`wallet=$(mktemp 2>/dev/null)`
to create the temp files in the RAM and reduce the SDcard wear.
Also this way they are flushed when powered off.
For extra safety can shred and remove the tempfiles once used.

rootzoll, 2021-03-08:
Is there anything more todo on the raspiblitz or do you just wanted to note this change as a FYI for the v1.7 release?

openoms, 2021-03-08:
I think this should be applied in RaspiBlitz as well. Will do some test and open a PR.

rootzoll, 2021-03-16:
#2058 got merged - will be part of v1.7 release - but keep open for final testing
- recheck password changes
- fresh node setup

rootzoll, 2021-04-16:
There are still some liitle things that can be moved to RAM ... pushing this for re-interation to v1.7.1

rootzoll, 2021-09-16:
The enw setup process now makes good use of the RAM disk. Closing issue.

closed 2021-09-16 by rootzoll