raspiblitz/raspiblitz #2037: build_sdcard.sh - !!! BUILD FAILED --> Was not able to install bitcoind version(0.21.0)

Neo4h0stKey, 2021-03-05:
Hello, if I want to install the Raspiblitz with "sudo bash build_sdcard.sh I get the error message when installing the bitcoin.org node and I don't know why
Can you help me?
the whole thing runs into one:
IntelNUC-Proxmox- (VM) RaspberryPI OS for x86

`2021-03-05 09:52:33 (25.6 MB/s) - ‘laanwj-releases.asc.1’ saved [32242/32242]

gpg: WARNING: no command supplied.  Trying to guess what you mean ...
pub   rsa4096 2015-06-24 [SC] [expires: 2022-02-10]
      01EA5486DE18A882D4C2684590C8019E36C2E964
uid           Wladimir J. van der Laan (Bitcoin Core binary release signing key) <laanwj@gmail.com>
gpg: key 90C8019E36C2E964: 51 signatures not checked due to missing keys
gpg: key 90C8019E36C2E964: "Wladimir J. van der Laan (Bitcoin Core binary release signing key) <laanwj@gmail.com>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
--2021-03-05 09:52:33--  https://bitcoin.org/bin/bitcoin-core-0.21.0/SHA256SUMS.asc
Resolving bitcoin.org (bitcoin.org)... 138.68.248.245
Connecting to bitcoin.org (bitcoin.org)|138.68.248.245|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1769 (1.7K) [application/octet-stream]
Saving to: ‘SHA256SUMS.asc.1’

SHA256SUMS.asc.1          100%[=====================================>]   1.73K  --.-KB/s    in 0s

2021-03-05 09:52:33 (68.9 MB/s) - ‘SHA256SUMS.asc.1’ saved [1769/1769]

goodSignature(1)
correctKey(1)

****************************************
OK --> BITCOIN MANIFEST IS CORRECT
****************************************

*** BITCOIN v0.21.0 for x86_64-linux-gnu ***

****************************************
OK --> VERIFIED BITCOIN CHECKSUM CORRECT
****************************************

bitcoin-0.21.0/
bitcoin-0.21.0/bin/
bitcoin-0.21.0/bin/bitcoin-cli
bitcoin-0.21.0/bin/bitcoind
bitcoin-0.21.0/bin/bitcoin-qt
bitcoin-0.21.0/bin/bitcoin-tx
bitcoin-0.21.0/bin/bitcoin-wallet
bitcoin-0.21.0/bin/test_bitcoin
bitcoin-0.21.0/include/
bitcoin-0.21.0/include/bitcoinconsensus.h
bitcoin-0.21.0/lib/
bitcoin-0.21.0/lib/libbitcoinconsensus.so
bitcoin-0.21.0/lib/libbitcoinconsensus.so.0
bitcoin-0.21.0/lib/libbitcoinconsensus.so.0.0.0
bitcoin-0.21.0/README.md
bitcoin-0.21.0/share/
bitcoin-0.21.0/share/man/
bitcoin-0.21.0/share/man/man1/
bitcoin-0.21.0/share/man/man1/bitcoin-cli.1
bitcoin-0.21.0/share/man/man1/bitcoind.1
bitcoin-0.21.0/share/man/man1/bitcoin-qt.1
bitcoin-0.21.0/share/man/man1/bitcoin-tx.1
bitcoin-0.21.0/share/man/man1/bitcoin-wallet.1
sudo: unable to execute /usr/local/bin/bitcoind: No such file or directory

!!! BUILD FAILED --> Was not able to install bitcoind version(0.21.0)
`

openoms, 2021-03-05:
Not sure if that is cause of the issue, but clearly you have some files left around from previous attempts:
```
laanwj-releases.asc.1
SHA256SUMS.asc.1
bitcoin-0.21.0/share/man/man1/
bitcoin-0.21.0/share/man/man1/bitcoin-cli.1
bitcoin-0.21.0/share/man/man1/bitcoind.1
bitcoin-0.21.0/share/man/man1/bitcoin-qt.1
bitcoin-0.21.0/share/man/man1/bitcoin-tx.1
bitcoin-0.21.0/share/man/man1/bitcoin-wallet.1
```
clean the directories and try again. 
Use the v1.6 branch for the latest scripts with:
`sudo bash build_sdcard.sh v1.6`

nyxnor, 2021-04-18:
@Neo4h0stKey Pls close this if solved.

closed 2021-04-29 by Neo4h0stKey