raspiblitz/raspiblitz #2038: Enable Wiki

keblek, 2021-03-05:
```
% wc -l README.md
1150 README.md
```
The readme is very very long but it has good information. 

It needs to be split up, the best way to do it is by enabling the wiki on GitHub and splitting up the information. 

This will help attract new users since they will see a short and concise readme that explains the project and how to get started.

Please enable the wiki @rootzoll so that I can start this.

Ben10Brasileiro, 2021-03-07:
I agree would be valuable to enable the wiki, I can help @keblek if it is done.

keblek, 2021-03-07:
Thank you @Ben10Brasileiro,
One thing I do need help with and I don't know if the maintainers can help, but deciding what to keep on the README.

Im thinking;
- RaspiBlitz
- Feature Overview
- Time Estimate to Set Up a RaspiBlitz
- Hardware Needed (with pricing of the standard and ready to go link)
- Support
- Community Development 

Then move the rest to the wiki, including the FAQ and WORKSHOP.

keblek, 2021-03-07:
In the meantime I've made this https://github.com/keblek/raspiblitzwiki

rootzoll, 2021-03-08:
Thanks for the input. @openoms we had some discussion about improving documentation in the past. Maybe starting with the github wiki would be first ay to start structuring it a bit more?

openoms, 2021-03-08:
Yes, it would be good to not need to merge PRs to improve the documentation.
Also agree that more structure would make it easier to digest.

keblek, 2021-03-08:
@rootzoll  if you could enable it, I can start making commits Settings in the repo->Options->Wikis

Then once it looks good enough you can update the main repo and add a note that documentation is in the wiki

Ben10Brasileiro, 2021-03-08:
Altough I understand there is a rush for you to update the wiki, if you could wait sometime causeit is faster if I commit in your repo (keblek) and a definitive (completed) wiki is commited. Or else me and you could make changes to the official repo that wont match. After it is a full written wiki, I think would be the best, but you choose.

I saw your repo but didnt have time to write a word, just want to highlight that it is best for the "Downloading software" and "Boot process" to be on the main short readme, as it is faster for everybody to just enter the repo and download the image, instead of clicking on another file.

Ben10Brasileiro, 2021-03-08:
Oh, I dindt see this. Sorry Keblek, if there is no need to PR for Documentation, than you are right.

Atenciosamente,

Benjamin Grande

Assessor

CALC Administração Judicial e Perícias

(43) 3026-5555 www.calc.com.br<http://www.calc.com.br>

rootzoll, 2021-04-16:
OK I moved this one step back to v1.7.1 (todo after v1.7 release) because it can make sense to combine adding a wiki with moving the raspiblitz repo from my personal user account to an organziation account - see:  #2184

keblek, 2021-04-16:
sounds good!

rootzoll, 2021-08-24:
Sorry v1.7.1 got already so big I move all those organizational changes up to v1.8

md-weber, 2022-06-29:
I would like to help here, could we already enable the Wiki on the [new repo](https://github.com/raspiblitz/raspiblitz.github.io) so that we could already work on it? 

Further reading: https://github.com/rootzoll/raspiblitz/issues/2184#issuecomment-821091983

rootzoll, 2022-06-30:
I am not sure about the process of moving a guthub repo yet - it may be that if you now start the wiki at the new repo and then I move the old to the new (to also keep stars & followers on the project) it might delete the other one - not sure.

But the wiki on this repo is open is you want to give it a try.

cstenglein, 2023-06-07:
Wiki has been enabled, so can be closed for now.

closed 2023-06-07 by cstenglein