raspiblitz/raspiblitz #2039: lnrpc error - local network connection refused causing reboot loop

jeremy-op, 2021-03-05:
PRECEDING ISSUE: I experienced an unknown error similar to https://github.com/rootzoll/raspiblitz/issues/1618. However, after several attempts to access, a final reboot resulted in ping/nmap not revealing host. This occurred soon after disabling lnbits and sphinx relay server following several weeks of nominal performance. I followed the issue and solutions, i.e. fan shim removal, etc, to no effect.

I had enabled/disabled various services and options: rtl, sphinx relay server (including an unsuccessful attempt to establish an IP2TOR Bridge), lightning loop, lightning pool, lnbits, static channel backup on usb drive, accept keysend

I had generated a recovery file immediately prior to the error.

I have reprovisioned 1.6.3 using the same password set as the original install, and installed recovery file via dialogue.

After complete chain sync, the raspiblitz goes into a reboot loop. 

The pi only remains open to my session briefly, so I was unable to copy current configs for bitcoin and lnd to determine if there is some rpc/password mismatch

Additionally, blk01567.dat seems unable to open, leading to a db error, unable to copy over a good file due to the reboot issue.

Unable to run 'systemctl daemon-reload' as logs suggest due to reboot issue.

Bootstrap update is again generating the same output as issue 1618 above.

I have reprovisioned a second time, with the same results.

I/O error requiring ssd replacement suspected, but just in case I wanted to post the issue for review.

Logs as follows:

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.6.3
chainnetwork: bitcoin / main
 17:03:19 up 2 min,  2 users,  load average: 1.91, 0.98, 0.38

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
Warning: The unit file, source configuration file or drop-ins of bitcoind.service changed on disk. Run 'systemctl daemon-reload' to reload units.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Fri 2021-03-05 17:00:59 GMT, end at Fri 2021-03-05 17:03:19 GMT. --
-- No entries --

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2021-03-05T16:42:50Z init message: Loading banlist...
2021-03-05T16:42:50Z Using /16 prefix for IP bucketing
2021-03-05T16:42:50Z Cache configuration:
2021-03-05T16:42:50Z * Using 2.0 MiB for block index database
2021-03-05T16:42:50Z * Using 8.0 MiB for chain state database
2021-03-05T16:42:50Z * Using 1014.0 MiB for in-memory UTXO set (plus up to 286.1 MiB of unused mempool space)
2021-03-05T16:42:50Z init message: Loading block index...
2021-03-05T16:42:50Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2021-03-05T16:42:50Z Opened LevelDB successfully
2021-03-05T16:42:50Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000
2021-03-05T16:43:03Z LoadBlockIndexDB: last block file = 2417
2021-03-05T16:43:03Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=56, size=72170298, heights=666095...667120, time=2021-01-15...2021-01-22)
2021-03-05T16:43:03Z Checking all blk files are present...
2021-03-05T16:43:03Z Unable to open file /mnt/hdd/bitcoin/blocks/blk01567.dat
2021-03-05T16:43:03Z : Error loading block database.
Please restart with -reindex or -reindex-chainstate to recover.
2021-03-05T16:43:03Z Aborted block database rebuild. Exiting.
2021-03-05T16:43:03Z Shutdown: In progress...
2021-03-05T16:43:03Z scheduler thread exit
2021-03-05T16:43:03Z Shutdown: done

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
Warning: The unit file, source configuration file or drop-ins of lnd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
[... 76 lines cut ...]
activated=0
connected=0
localip=''

*** HARDWARE TEST RESULTS ***
raspberryPi=4
powerFAIL=0
powerWARN=0
powerMIN='870000 microVolt'
tempFAIL=0
tempWARN=0
tempMAX='4300 centiGrad'

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.x.x'
tempCelsius='42.8'
uptime=128
upsStatus='OFF'
startcountBlockchain=0
bitcoinActive=0
startcountLightning=0
lndActive=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=0

Ben10Brasileiro, 2021-03-07:
Use the recommended SSD enclosure. Others can be faulty.
https://github.com/rootzoll/raspiblitz#package-standard-around-250-usd

rootzoll, 2021-03-08:
@jeremy-op which ssd casing are you using? Is it that one from the shopping list?

jeremy-op, 2021-03-08:
I am using a 1tb Verbatim external ssd

jeremy-op, 2021-03-08:
https://www.amazon.com/Verbatim-1TB-External-Hard-Drive/dp/B005AT5V00

rootzoll, 2021-03-08:
So this is an HDD not an SSD .. I dont have experience with that brand. But you could rule out the problem when you run the HDD on a USB2 port (instead on the blue USB3 ports) of the RaspberryPi4. USB2 is much slower - but because you use an HDD anyway the slowdown should not be too much.

jeremy-op, 2021-03-08:
Yes, sorry - I meant hdd. I see now I typed it incorrectly.

jeremy-op, 2021-03-08:
Running over USB2 caused the raspiblitz to recover previous config, prompt for new password 'a', force reboot, on reboot 'blockchain info' loading screen displays while the same issue occurs. Logs as follows:

***********************************************
Running RaspiBlitz Bootstrap 1.6.3
Mon  8 Mar 18:31:15 GMT 2021
***********************************************
# blitz.lcd.sh check-repair
# --> old kernel detected - no need to update LCD drivers.
error='unkown command'
isMounted: 0
HDD is there but not AutoMounted yet - checking Setup
Temp mounting data drive (sda)
Refreshing links between directories/drives ..
HDD contains already a configuration: 1
Found existing configuration
Found valid configuration
Calling Data Migration ..
STARTED Migration/Init
prechecks OK
default values OK
Checking old bitcoin.conf ...
check issue #217 -> ok rpcallow exists
check issue #950 -> ok ^rpcport does not exist
check issue #950 -> ok ^rpcbind does not exist
check issue #217 -> ok main.rpcport exists
check issue #217 -> ok main.rpcbind exists
check issue #950 -> ok test.rpcport exists
check issue #950 -> ok test.rpcbind exists
Version Code: 1.6.3
Version Data: 1.6.3
OK - version of config data is up to date
END Migration/Init
Calling Provisioning ..
STARTED Provisioning from preset config file
CHECK IF SD CARD NEEDS EXPANSION
rootPartition(mmcblk0p2)
rootPartitionBytes(63591940096)
### CHECKING ROOT PARTITION SIZE ###
Size in Bytes is: 63591940096 bytes on (mmcblk0p2)
Size looks good. Bigger than 8192000000 byte disk is used.
### BASIC SYSTEM SETTINGS ###
Setting lightning alias: XXXXXX
Old SSH PubKey exists on HDD > copy them HDD to SD card for next start
Copy HDD content for user admin
mkdir: cannot create directory ‘/home/admin/.bitcoin’: File exists
mkdir: cannot create directory ‘/home/admin/.lnd’: File exists
cp: '/mnt/hdd/lnd/lnd.conf' and '/home/admin/.lnd/lnd.conf' are the same file
cp: '/mnt/hdd/lnd/tls.cert' and '/home/admin/.lnd/tls.cert' are the same file
mkdir: cannot create directory ‘/home/admin/.lnd/data’: File exists
*** Make backup of LND directory
OK Backup is valid.

installing bash completion for bitcoin-cli and lncli
--2021-03-08 18:33:05--  https://raw.githubusercontent.com/bitcoin/bitcoin/master/contrib/bitcoin-cli.bash-completion
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.110.133, 185.199.109.133, 185.199.111.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4247 (4.1K) [text/plain]
Saving to: ‘bitcoin-cli.bash-completion’

     0K ....                                                  100% 9.83M=0s

2021-03-08 18:33:06 (9.83 MB/s) - ‘bitcoin-cli.bash-completion’ saved [4247/4247]

--2021-03-08 18:33:06--  https://raw.githubusercontent.com/lightningnetwork/lnd/master/contrib/lncli.bash-completion
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.108.133, 185.199.110.133, 185.199.109.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1498 (1.5K) [text/plain]
Saving to: ‘lncli.bash-completion’
[... 176 lines cut ...]
Running the custom install script ..
'/mnt/hdd/app-data/custom-installs.sh' -> '/home/admin/custom-install.sh'
There are no custom user installs so far.
Done

*** Replay backup of LND conf/tls
Copying TLS ...
On next final restart admin creds will be updated by _boostrap.sh
DONE

*** Repair Bitcioin Conf (if needed)
Aligning lnd.conf & bitcoin.conf
Not setting local network hostname
Prepare fstab for permanent data drive mounting ..
datadisk --> sda
datapartition --> sda1
# unmounting all drives
# waiting until uuid gets available
# mount /mnt/hdd
# waiting until mounting is active
# OK - fstab updated for EXT4 layout
DONE - Give raspi some cool off time after hard building .... 5 secs sleep
END Provisioning
rebooting
shutdown in 1min

rootzoll, 2021-03-08:
We recommedn the use of SSDs nowadays because HDDs on a raspberrypi tend to get into data corruption problems. So I would recommend to get a SSD - best would also to stick close to the shopping list because some SSDs are too power hungry also for the RaspberryPi and as mentioned some SSD casings also make problems. Those mentioned in the shopping list we had the best results so far.

If you had funds/channels on the LND I would recommend to get to the terminal and export an LND-Rescue-File and use that then again on a fresh setup with the new SSD. This chapter in the FAQ should be a good starting point: https://github.com/rootzoll/raspiblitz/blob/v1.6/FAQ.md#how-can-i-recover-my-coins-from-a-failing-raspiblitz

jeremy-op, 2021-03-08:
Ok, thank you.

closed 2021-03-08 by jeremy-op