raspiblitz/raspiblitz #2041: lndconnect install/build failure

brandonshope, 2021-03-07:
creating install info ...
# OK - file exists: /home/admin/raspiblitz/pictures/app_zap.png
fbi: no process found
Error: no such file ":https://github.com/rootzoll/raspiblitz/issues/1001#issuecomment-634580257"
# yes-button -> TOR
# Installing lndconnect..
Check Framework: Go

Installed go version go1.13.3 linux/arm

go: finding github.com/glendc/go-external-ip latest
go: finding github.com/skip2/go-qrcode latest
go: finding github.com/Baozisoftware/qrcode-terminal-go latest
HEAD is now at 82d7103 Merge pull request #15 from Kixunil/patch-1
go get ./...
go: github.com/rootzoll/lndconnect: looping trying to add package
make: *** [Makefile:4: dep] Error 1

!!!!!!!!!!!!!!!!!!!!!!!!!!
FAIL: Was not able to install/build lndconnect
Retry later or report to developers with logs above.
/home/admin/config.scripts/bonus.lndconnect.sh: line 55: lndconnect: command not found
!!!!!!!!!!!!!!!!!!!!!!!!!!

rootzoll, 2021-03-08:
So lndconnect install is not working on your node - this is tested with that line: https://github.com/rootzoll/raspiblitz/blob/9ae1a2d9bfdd4d644ed103ef69442dd6b31899f7/home.admin/config.scripts/bonus.lndconnect.sh#L55

@brandonshope on which action did you get the error? Did you retried the install?

nyxnor, 2021-04-18:
Close this pls.

closed 2021-04-19 by rootzoll