raspiblitz/raspiblitz #2048: Detect if a SSD casing has UAS issues

rootzoll, 2021-03-11:
see https://github.com/getumbrel/umbrel/pull/580

rootzoll, 2021-03-26:
the umbrel way was not working to detect the PID & VID ... but now merged an alternative detection

rootzoll, 2021-03-26:
looks good so far, will be part of v1.7RC1 - leaving open for final testing

rootzoll, 2021-03-27:
@openoms do you now a ssd casing I can order from amazon THAT HAS the UASP problem so I can test this for the final release?

openoms, 2021-03-27:
haven't got a link, but basically all my old (pre RPi4) USB adapters had the issue (except the X820 board): https://twitter.com/openoms/status/1158876381306064897

rootzoll, 2021-03-27:
TODO: the auto-reboot was deactivated for the RC1 - TODO: check to prevent reboot-loops on update for final release

rootzoll, 2021-03-31:
BTW the USAP could be one reason why the RaspiBlitz is faster in syncing IBD in the past then umbrel that deactivate USAP by default (like the idital idea of this issue).

I would improve on this that if the SSD-Case Model is not on a "whitelist" that then USAP should be deactivated. So for v1.7 I will try if we can test against the model ID of the UGREEN and the X820 board ... if its any other the USAP should be deactivated.

rootzoll, 2021-04-18:
OK I tested now with two cases that were reported as prolematic before - they get marked in /boot/cmdline.txt on install and sync thru so far good. Will be part of v1.7 .. closing issue.

closed 2021-04-18 by rootzoll