raspiblitz/raspiblitz #2050: Trouble for Sync RaspiBlitz

titidu35, 2021-03-12:
Hello everybody, i'm a french user and i have some problem to sync my RaspiBlitz. First time config with Tor and at final without Tor.
I don't understand what it block, i take IP fixe on my Box and open port 8333...

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.6.3
chainnetwork: bitcoin / main
 11:03:43 up 23 min,  2 users,  load average: 0.20, 0.19, 0.20

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-03-12 10:44:02 GMT; 19min ago
  Process: 8181 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 8186 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 8228 (bitcoind)
    Tasks: 17 (limit: 4915)
   Memory: 43.5M
   CGroup: /system.slice/bitcoind.service
           └─8228 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.c…

Mar 12 10:44:02 raspberrypi systemd[1]: Starting Bitcoin daemon...
Mar 12 10:44:02 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Fri 2021-03-12 10:39:33 GMT, end at Fri 2021-03-12 11:03:44 GMT. --
Mar 12 10:44:02 raspberrypi systemd[1]: Starting Bitcoin daemon...
Mar 12 10:44:02 raspberrypi systemd[1]: Started Bitcoin daemon.
1615545840 ERROR: Error found in Logs

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2021-03-12T10:44:02Z init message: Loading P2P addresses...
2021-03-12T10:44:02Z ERROR: DeserializeFileDB: Failed to open file /mnt/hdd/bitcoin/peers.dat
2021-03-12T10:44:02Z Invalid or missing peers.dat; recreating
2021-03-12T10:44:02Z init message: Starting network threads...
2021-03-12T10:44:02Z net thread start
2021-03-12T10:44:02Z dnsseed thread start
2021-03-12T10:44:02Z Loading addresses from DNS seed dnsseed.emzy.de
2021-03-12T10:44:02Z msghand thread start
2021-03-12T10:44:02Z init message: Done loading
2021-03-12T10:44:02Z opencon thread start
2021-03-12T10:44:02Z addcon thread start
2021-03-12T10:44:03Z Loading addresses from DNS seed seed.bitcoin.jonasschnelli.ch
2021-03-12T10:44:03Z Loading addresses from DNS seed dnsseed.bitcoin.dashjr.org
2021-03-12T10:44:03Z Loading addresses from DNS seed dnsseed.bluematt.me
2021-03-12T10:44:04Z Loading addresses from DNS seed seed.bitcoin.sipa.be
2021-03-12T10:44:04Z Loading addresses from DNS seed seed.bitcoin.sprovoost.nl
2021-03-12T10:44:04Z Loading addresses from DNS seed seed.bitcoinstats.com
2021-03-12T10:44:04Z Loading addresses from DNS seed seed.btc.petertodd.org
2021-03-12T10:44:04Z 277 addresses found from DNS seeds
2021-03-12T10:44:04Z dnsseed thread exit

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-03-12 10:44:02 GMT; 19min ago
  Process: 8229 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 8231 (lnd)
    Tasks: 9 (limit: 4915)
   Memory: 574.1M
   CGroup: /system.slice/lnd.service
           └─8231 /usr/local/bin/lnd --externalip=127.0.0.1:9735

Mar 12 10:44:02 raspberrypi systemd[1]: Starting LND Lightning Daemon...
Mar 12 10:44:02 raspberrypi systemd[1]: Started LND Lightning Daemon.

*** LAST LND ERROR LOGS ***
[... 83 lines cut ...]
tempWARN=0
tempMAX='5200 centiGrad'

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.1.21'
tempCelsius='47.7'
uptime=1439
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=0
initialSync=1
syncProgress=0.00 
startcountLightning=1
lndActive=1
walletLocked=0
lndPeers=0
syncedToChain=0
scanTimestamp=1231006505
scanDate='Sat  3 Jan 18:15:05 GMT 2009'
scanProgress=.01
lndRPCReady=1
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=1

titidu35, 2021-03-12:
no Blockchain progress and no Lightning progress (0%, 0 peer)

nyxnor, 2021-04-18:
@titidu35 Solved? If yes, close this pls.

nyxnor, 2021-05-17:
@rootzoll Please close this

closed 2021-05-18 by rootzoll