raspiblitz/raspiblitz #2051: Fee-PerMil unit is confusing or wrong

Svobodan, 2021-03-13:
**Issue:**
When listing channels with "lnchannels" option in RaspiBlitz menu the description says:

> Fee-PerMil -> amount based fee (millisatoshis per 1 satoshi) on forwarding

But then the table shows value that is 1000 times higher than it should. For instance the default fee rate value after opening a channel is 1 meaning one milionth, not one thousandth as suggested. If value is set to 10000, then fee rate is 1%.

Or perhaps the intention was to say "1 permil of a millisatoshi per satoshi"? This would be confusing for human as it combines fractions that include prefixes with percentages.

**Suggestion:**

1. Change values (divide by 1000) in the table to match PerMil (‰)
or
2. Change values to percent - most familiar to humans but too many decimal places
or
3. Change description to:
> Fee Rate -> amount based fee (1 satoshi per million satoshi) on forwarding

**Background:**
I remember I was confused the first time seeing this and I just assumed that permil means per million but in fact it means per thousand. Today I was remindend of this when responding to someone regarding RTL where I describe what units should be used.

Svobodan, 2021-03-13:
[https://github.com/Ride-The-Lightning/RTL/issues/621#issuecomment-798298671](url)

rootzoll, 2021-03-14:
Thanks I will take a look for the next release.

Svobodan, 2021-03-14:
This is what they have on 1ml.com. I find that quite easy to read.

[image]

openoms, 2021-03-15:
Fee calculation has many standards unfortunately:

[image]
[image]

closed 2024-08-06 by rootzoll