raspiblitz/raspiblitz #2052: "The bitcoind service is NOT Running!" [Blitzversion 1.6.3]

techric, 2021-03-13:
I'm trying to configure a new RaspiBlitz (v1.6.3) for a Bitcoin Lightning Node

While trying to sync with the Bitcoin blockchain, I get an error message that reads: 

**_"The bitcoind service is NOT RUNNING! Error found in Logs Login for more details and options: ssh admin@[My IP Address] -> Password A"_**

**Here's the error log …**

***************************************************************
[*](url) RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.6.3
chainnetwork: bitcoin / main
 21:49:45 up 40 min,  5 users,  load average: 0.21, 0.22, 0.29

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2021-03-13 21:49:43 GMT; 2s ago
  Process: 9640 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 9642 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=1/FAILURE)

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Sat 2021-03-13 21:08:54 GMT, end at Sat 2021-03-13 21:49:45 GMT. --
Mar 13 21:49:43 raspberrypi systemd[1]: bitcoind.service: Service RestartSec=30s expired, scheduling restart.
Mar 13 21:49:43 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 67.
Mar 13 21:49:43 raspberrypi systemd[1]: Stopped Bitcoin daemon.
Mar 13 21:49:43 raspberrypi systemd[1]: Starting Bitcoin daemon...
Mar 13 21:49:43 raspberrypi bitcoind[9642]: Error: Cannot write to data directory '/home/bitcoin/.bitcoin'; check permissions.
Mar 13 21:49:43 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
Mar 13 21:49:43 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Mar 13 21:49:43 raspberrypi systemd[1]: Failed to start Bitcoin daemon.
1615672157 ERROR: Error found in Logs
1615672178 ERROR: Error found in Logs

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2021-03-13T21:14:20Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2021-03-13T21:14:20Z HTTP: starting 4 worker threads
2021-03-13T21:14:20Z init message: Loading banlist...
2021-03-13T21:14:20Z Using /16 prefix for IP bucketing
2021-03-13T21:14:20Z Cache configuration:
2021-03-13T21:14:20Z * Using 2.0 MiB for block index database
2021-03-13T21:14:20Z * Using 8.0 MiB for chain state database
2021-03-13T21:14:20Z * Using 1014.0 MiB for in-memory UTXO set (plus up to 286.1 MiB of unused mempool space)
2021-03-13T21:14:20Z init message: Loading block index...
2021-03-13T21:14:20Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2021-03-13T21:14:20Z Opened LevelDB successfully
2021-03-13T21:14:20Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000
2021-03-13T21:14:32Z LoadBlockIndexDB: last block file = 9
2021-03-13T21:14:32Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=2445, size=64053405, heights=176052...178540, time=2012-04-17...2012-05-04)
2021-03-13T21:14:32Z Checking all blk files are present...
2021-03-13T21:14:32Z Opening LevelDB in /mnt/hdd/bitcoin/chainstate
2021-03-13T21:14:33Z Opened LevelDB successfully
2021-03-13T21:14:33Z Using obfuscation key for /mnt/hdd/bitcoin/chainstate: 7ed0124bee95a2f3
2021-03-13T21:14:33Z Loaded best chain: hashBestChain=000000000000018153b94e54712a8692c1db3984f681f4fa5d9bb73ba676465b height=178479 date=2012-05-03T21:24:52Z progress=0.004851
2021-03-13T21:14:33Z init message: Rewinding blocks...

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2021-03-13 21:14:20 GMT; 35min ago
  Process: 4671 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 4673 (lnd)
    Tasks: 9 (limit: 4915)
   Memory: 27.4M
   CGroup: /system.slice/lnd.service
           └─4673 /usr/local/bin/lnd --externalip=68.206.97.22:9735

[... 91 lines cut ...]
lndRPCReady=1
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=0

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

admin@raspberrypi:~ $ sudo tail -f /mnt/hdd/bitcoin/debug.log
2021-03-13T21:14:20Z Opened LevelDB successfully
2021-03-13T21:14:20Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000
2021-03-13T21:14:32Z LoadBlockIndexDB: last block file = 9
2021-03-13T21:14:32Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=2445, size=64053405, heights=176052...178540, time=2012-04-17...2012-05-04)
2021-03-13T21:14:32Z Checking all blk files are present...
2021-03-13T21:14:32Z Opening LevelDB in /mnt/hdd/bitcoin/chainstate
2021-03-13T21:14:33Z Opened LevelDB successfully
2021-03-13T21:14:33Z Using obfuscation key for /mnt/hdd/bitcoin/chainstate: 7ed0124bee95a2f3
2021-03-13T21:14:33Z Loaded best chain: hashBestChain=000000000000018153b94e54712a8692c1db3984f681f4fa5d9bb73ba676465b height=178479 date=2012-05-03T21:24:52Z progress=0.004851
2021-03-13T21:14:33Z init message: Rewinding blocks...

[image]
[image]
[image]

openoms, 2021-03-13:
From this:
`Error: Cannot write to data directory '/home/bitcoin/.bitcoin`
It is likely to be a hardware failure either your SDcard failing (so need changing)
or the disk.
Make sure you use one of the tested and recommended USB-disk adapters because incompatibility can cause all kind of errors.

nyxnor, 2021-04-18:
@techric Solved? If yes, close this issue pls.

techric, 2021-04-18:
Issued solved. 

Node is up and running. 

Thanks for all of your help.

closed 2021-04-18 by techric

paranoidnakamoto, 2021-05-19:
Very interesting problem, you can close this problem after reading.
If you want to resolve this issue, you must connect your SSD to external power, this is possible?
After connecting the SSD with an external SSD converter and power supply + USB 3.0, it is suddenly possible to execute this SSD and write without problems.

paranoidnakamoto, 2021-05-19:
No fixed solution but worth an attempt ..When delivery chain by Amazon does not cover can also use a USB to SATA Converter. Important is external power.
Problem seems to be apparently with others that the power storage is not good enough.