raspiblitz/raspiblitz #2053: scanProgress=99.99

exfrog, 2021-03-14:
Node was off for over 6 months.  re-sync only took less than 1 day, and the pi looks like everything is working correctly, but RTL and lncli both give me 'server is still in the process of starting' if i try to add a peer or force close a channel.

only thing that stood out to me was at the end of debug - scanProgress=99.99
its been like this for over a day. do i just need to wait longer? is there another issue?
anything stand out in this log?   thanks for your help

https://termbin.com/zyr2
\/
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.6.3
chainnetwork: bitcoin / main
 10:43:33 up  1:53,  3 users,  load average: 3.56, 3.99, 3.72

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-03-14 08:52:45 GMT; 1h 50min ago
  Process: 9080 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 9088 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 9148 (bitcoind)
    Tasks: 18 (limit: 4915)
   Memory: 5.1G
   CGroup: /system.slice/bitcoind.service
           └─9148 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Mar 14 08:52:45 raspberrypi systemd[1]: Starting Bitcoin daemon...
Mar 14 08:52:45 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Sun 2021-03-14 08:49:43 GMT, end at Sun 2021-03-14 10:43:33 GMT. --
Mar 14 08:52:45 raspberrypi systemd[1]: Starting Bitcoin daemon...
Mar 14 08:52:45 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2021-03-14T09:35:01Z New outbound peer connected: version: 70015, blocks=674564, peer=32 (block-relay)
2021-03-14T09:35:30Z UpdateTip: new best=00000000000000000003bbde15bfb1a38748b056ab53cba5a2d12c6811457a54 height=674565 version=0x20800000 log2_work=92.734287 tx=624679646 date='2021-03-14T09:34:40Z' progress=1.000000 cache=48.9MiB(442121txo) warning='72 of last 100 blocks have unexpected version'
2021-03-14T09:35:30Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2021-03-14T09:39:16Z UpdateTip: new best=0000000000000000000789495e84e34bf927482ddd3f8819d0f8c84cc5dd022a height=674566 version=0x20c00000 log2_work=92.734303 tx=624680907 date='2021-03-14T09:37:50Z' progress=1.000000 cache=54.7MiB(479409txo) warning='72 of last 100 blocks have unexpected version'
2021-03-14T09:39:16Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2021-03-14T09:40:23Z Pre-allocating up to position 0xc00000 in rev02491.dat
2021-03-14T09:40:23Z UpdateTip: new best=0000000000000000000503c50ce44e2ff62492d7605590168964e4587a1eb09d height=674567 version=0x20400000 log2_work=92.734319 tx=624682233 date='2021-03-14T09:39:01Z' progress=1.000000 cache=56.2MiB(493572txo) warning='73 of last 100 blocks have unexpected version'
2021-03-14T09:40:23Z BlockUntilSyncedToCurrentChain: txindex is catching up on block notifications
2021-03-14T09:44:49Z ping timeout: 1200.021241s
2021-03-14T09:45:10Z New outbound peer connected: version: 70015, blocks=674567, peer=40 (full-relay)
2021-03-14T09:55:10Z Imported mempool transactions from disk: 21346 succeeded, 676 failed, 0 expired, 1109 already there
2021-03-14T09:57:05Z UpdateTip: new best=00000000000000000009353dd85f985120a7d2d9e09f2735328c0a56c6994d06 height=674568 version=0x20000000 log2_work=92.734335 tx=624684678 date='2021-03-14T09:56:47Z' progress=1.000000 cache=69.3MiB(615596txo) warning='72 of last 100 blocks have unexpected version'
2021-03-14T10:19:45Z UpdateTip: new best=00000000000000000001768ef9ad710303f3af1c2bb1f7ffc59b7b5bf27e121b height=674569 version=0x2fffe000 log2_work=92.734351 tx=624686604 date='2021-03-14T10:19:33Z' progress=1.000000 cache=71.3MiB(634866txo) warning='72 of last 100 blocks have unexpected version'
2021-03-14T10:25:28Z UpdateTip: new best=0000000000000000000450ff41c5ccdd13ccd1a8b90966a50f5fc7c6acfd5934 height=674570 version=0x27ffe000 log2_work=92.734367 tx=624688487 date='2021-03-14T10:25:16Z' progress=1.000000 cache=72.0MiB(641505txo) warning='73 of last 100 blocks have unexpected version'
2021-03-14T10:27:17Z New outbound peer connected: version: 70015, blocks=674570, peer=64 (full-relay)
2021-03-14T10:34:18Z New outbound peer connected: version: 70015, blocks=674570, peer=67 (full-relay)
2021-03-14T10:36:38Z Pre-allocating up to position 0x6000000 in blk02491.dat
2021-03-14T10:36:39Z UpdateTip: new best=00000000000000000001fca7b9ad6e502797f962542fe804640cf568615caf76 height=674571 version=0x20800000 log2_work=92.734384 tx=624691023 date='2021-03-14T10:36:06Z' progress=1.000000 cache=73.2MiB(652584txo) warning='73 of last 100 blocks have unexpected version'
2021-03-14T10:37:35Z ThreadRPCServer incorrect password attempt from 127.0.0.1:37416
2021-03-14T10:39:05Z UpdateTip: new best=000000000000000000065cf0ac5e7b36c36e228a17b1093cd2630c9d52def5bc height=674572 version=0x20c00000 log2_work=92.7344 tx=624692060 date='2021-03-14T10:39:01Z' progress=1.000000 cache=73.8MiB(658424txo) warning='74 of last 100 blocks have unexpected version'

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2021-03-14 08:52:45 GMT; 1h 50min ago
  Process: 9154 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 9161 (lnd)
    Tasks: 12 (limit: 4915)
   Memory: 231.4M
   CGroup: /system.slice/lnd.service
           └─9161 /usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v3 --listen=127.0.0.1:9735 --accept-keysend

Mar 14 08:52:45 raspberrypi systemd[1]: Starting LND Lightning Daemon...
[... 112 lines cut ...]
Mar 14 10:43:15 raspberrypi electrs[9191]: 2021-03-14T10:43:15.230+00:00 - TRACE - [127.0.0.1:52012] joining ThreadId(394)
Mar 14 10:43:15 raspberrypi electrs[9191]: 2021-03-14T10:43:15.230+00:00 - INFO - [127.0.0.1:52012] connected peer
Mar 14 10:43:15 raspberrypi electrs[9191]: 2021-03-14T10:43:15.231+00:00 - TRACE - RPC Request("{\"jsonrpc\":\"2.0\",\"method\":\"server.ping\",\"params\":[],\"id\":\"electrs-check\"}\n")
Mar 14 10:43:16 raspberrypi electrs[9191]: 2021-03-14T10:43:16.792+00:00 - TRACE - RPC PeriodicUpdate
Mar 14 10:43:16 raspberrypi electrs[9191]: 2021-03-14T10:43:16.793+00:00 - TRACE - RPC PeriodicUpdate
Mar 14 10:43:17 raspberrypi electrs[9191]: 2021-03-14T10:43:17.238+00:00 - TRACE - RPC Done
Mar 14 10:43:17 raspberrypi electrs[9191]: 2021-03-14T10:43:17.238+00:00 - DEBUG - [127.0.0.1:52012] shutting down connection
Mar 14 10:43:17 raspberrypi electrs[9191]: 2021-03-14T10:43:17.238+00:00 - INFO - [127.0.0.1:52012] disconnected peer
Mar 14 10:43:22 raspberrypi electrs[9191]: 2021-03-14T10:43:22.454+00:00 - TRACE - RPC Request("{\"jsonrpc\":\"2.0\",\"method\":\"server.ping\",\"params\":[],\"id\":51}\n")
Mar 14 10:43:22 raspberrypi electrs[9191]: 2021-03-14T10:43:22.578+00:00 - TRACE - RPC Request("{\"jsonrpc\":\"2.0\",\"method\":\"server.ping\",\"params\":[],\"id\":557}\n")
Mar 14 10:43:23 raspberrypi electrs[9191]: 2021-03-14T10:43:23.771+00:00 - TRACE - RPC PeriodicUpdate
Mar 14 10:43:29 raspberrypi electrs[9191]: 2021-03-14T10:43:29.426+00:00 - TRACE - RPC PeriodicUpdate
Mar 14 10:43:32 raspberrypi electrs[9191]: 2021-03-14T10:43:32.582+00:00 - TRACE - RPC Request("{\"jsonrpc\":\"2.0\",\"method\":\"server.ping\",\"params\":[],\"id\":558}\n")

*** ElectRS Status ***
##### STATUS ELECTRS SERVICE
configured=1
serviceInstalled=1
serviceRunning=1
tipSynced=0
electrumResponding=1
initialSynced=1
localIP='10.1.1.217'
portTCP='50001'
localTCPPortActive=1

rootzoll, 2021-03-14:
I dont see anythink on the first look .. have you tried a basic retstart? login per SSH - then break out the info screen with CTRL+C and then in terminal type "restart"

exfrog, 2021-03-14:
yeah, done multiple restarts.  each time, the rpi screen reports its status in the startup process, then proceeds to the main screen where it does not indicate any problem, but still cannot perform any standard lncli or RTL actions.

other potentially related info:  
- started this whole process with a new blitz image on a new sd card, and have been super impressed with the systems ability to recover itself up to this point
- during initial re-sync, i flagged some options on, like electrs, mempool, etc.    and they installed and rebooted while the re-sync was still in progress....
- at some point, i upgraded from a pi3 to pi4. pretty sure everything was working on the pi4 before i had to move locations,  but hard to remember now...
- pi4 was setup with wired IP on .214  then moved to wireless IP on .217.   all forwarding is setup for .217  and all the services are showing up there correctly (spectre, block explorer etc etc)

Is there any way to see debug info on the 'scanProgress' value?    
Is scanProgress related to the ln channel graph, and is this necessary, or is there a way to bypass?

I've got about 2.5m sats locked up in 2 channels that were open and active when i had to move physical locations, so the end goal is re-connecting with these funds if possible.

Thanks again for any support!

exfrog, 2021-03-14:
after enough patience after reboot, lnd found and connected to piers, and my channels updated to reflect the remote closes that i assumed had occured.
seems like everything is working, although debug still reports scanProgress=99.99
closing issue
thanks!!

closed 2021-03-14 by exfrog