raspiblitz/raspiblitz #2054: Tor domain, torproject.org blocked in authocratic surveillance regime jurisdiction

pretyflaco, 2021-03-14:
During initial setup up of Raspiblitz v1.6.3, the blitz tries to fetch new packages from https://deb.torproject.org and fails because the domain is blocked by ISP (national regulation). I think raspiblitz can still fall back to the package version from the img-file, so not critical at the moment. If tor upgrade becomes necessary in the future, it could become a problem though if there's no mirror to fallback to. Maybe it would be possible to add alternate sources to fetch?

[image]

rootzoll, 2021-03-14:
The v1.7 will have "Tor by default" and already have it installed on sd build-script - so it will be part of the binary.

But I keep this issue open as if anybody has an idea how to tackle this problem in general .. because it should stay possible also for people to build the sd card image themselves in such regions.

NilByte, 2021-03-14:
I would try setting static DNS not provided by your ISP in your router.

Try these ones from Cloudflare and Google.

[image]

Ben10Brasileiro, 2021-03-16:
**BRIDGES**
Official code here on line 114
https://gitweb.torproject.org/torspec.git/tree/attic/bridges-spec.txt

https://bbs.archlinux.org/viewtopic.php?id=204312
https://tor.stackexchange.com/questions/3924/how-to-add-obfs3-bridges-in-torrc

I did not found how to connect to Tor Briges using torrc via the main Tor Project website, just how to run a bridge on the main pages https://community.torproject.org/relay/setup/bridge/debian-ubuntu/
https://2019.www.torproject.org/docs/bridges.html.en#RunningABridge

Kixunil, 2021-03-17:
FYI to my knowledge APT supports Tor but I never tried configuring it manually.

Ben10Brasileiro, 2021-03-17:
@Kixunil 
Here https://support.torproject.org/apt/apt-over-tor/
But they did not mention the code to include the signature. Here it is for Debian Stable:
```
curl -s --socks5-hostname 127.0.0.1:9050 http://sdscoq7snqtznauu.onion/torproject.org/A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89.asc | apt-key add
````

pretyflaco, 2021-03-20:
Thanks to the help and instructions of @Ben10Brasileiro the following steps allow for updating Tor packages on my blitz:

change to root
$ sudo su

Install the tor transport packages
$ apt install apt-transport-tor

Add bridges to torrc
$ nano /etc/tor/torrc

UseBridges 1
UpdateBridgesFromAuthority 1
bridge obfs4 ipAddress:443 fingerprintHere
bridge obfs4 ipAddress:443 fingerprintHere
bridge obfs4 ipAddress:443 fingerprintHere

Edit sources.list and replace the https:// with tor:// for tor packages. Sources.list looks as written below.
$ nano /etc/apt/sources.list

deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
deb tor://sdscoq7snqtznauu.onion/torproject.org buster main
#deb tor://deb.torproject.org/torproject.org buster main
#deb-src tor://deb.torproject.org/torproject.org buster main

Add the keys
$ torsocks wget -qO- http://sdscoq7snqtznauu.onion/torproject.org/A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89.asc | gpg --import
$ gpg --export A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89 | apt-key add -
$ apt install deb.torproject.org-keyring

Update the sources
$ apt update

Restart
$ restart

rootzoll, 2021-03-23:
Adding this issue now for v1.8 release milestone to be integrated to the sd build card.

rootzoll, 2021-03-23:
If some likes to turn this into a PullRequest that refactors the Tor install in the `build_sdcard.sh` script ..
https://github.com/rootzoll/raspiblitz/blob/158b73291f1b9f5826d61cb6a12f8907d41b0a14/build_sdcard.sh#L171

.. over the Lightning Hacksprint comming up this weekend, we can also put this already into the v1.7 release. here are the deatils on the event: https://wiki.fulmo.org/wiki/Lightning_HackSprint_March_2021

Ben10Brasileiro, 2021-03-23:
>If someone likes to turn this into a PullRequest that refactors the Tor install in the build_sdcard.sh script ..

I dont have all the abilities to correct this, but this is what need to be put into consideration:

### 1. If torproject.org domain is blocked:
* using a mirror (a lot of mirrors are outdated)
* apt over tor to be certain of getting the latest release

### 2. If they live on a country that bans use of tor:
* try to use bridges as fast as possible, when installing Tor as it will automatically start running. To avoid any tor connection being caught from the ISP, maybe [try this](https://askubuntu.com/questions/74061/install-packages-without-starting-background-processes-and-services) for the first install, add bridge and remove it.
* apt over tor so their ISP or GOV wont know what is being downloaded.

### I can see 3 options to be chosen from this:

bash build_sdcard.sh -t (Tor repo > Normal)
bash build_sdcard.sh -b (Tor repo + APToverTor + Bridges > torproject.org domain and Tor blocked)
bash build_sdcard.sh -m (Mirror > torproject.org domain blocked)

Kixunil, 2021-03-23:
Those SO suggestions are quite crappy. The cleanest way I know of is running `systemctl mask tor@default` before installation. `policy-rc.d` should be also quite good.

Ben10Brasileiro, 2021-03-23:
@Kixunil This right? The 3rd solution here https://askubuntu.com/questions/74061/install-packages-without-starting-background-processes-and-services on the mentioned link above
[image]

Kixunil, 2021-03-23:
Yes that one, but see the comments below it. Maybe non-issue for Raspiblitz, but better to keep it in mind.

Ben10Brasileiro, 2021-03-24:
Indeed @Kixunil Tried the `mask` now with Tor on a new system. Did gracefully, clean execution.
Im still editing this code (some bugs), but here it is. Search for "mask" and "unmask"
https://github.com/Ben10Brasileiro/raspiblitz/blob/patch-4/home.admin/config.scripts/internet.tor.sh

If the user dont have `UseBridges 1` on torrc, it will be prompted to him some questions. Torrrc dont have this line by default, so on first install it is ok, but after modifying it, if it is commented, adding bridges will be ignored. Did not find a way better to set this/

Kixunil, 2021-03-24:
Maybe you would be interested in my package called [`tor-hs-config-patch`](https://github.com/Kixunil/cryptoanarchy-deb-repo-builder/tree/master/tor-extras/assets), which does clean patching of Tor installation to enable reading parts configuration from `/etc/tot/torrc.d` and `/etc/tor/hidden-services.d`.

The advantages of this approach:

* User can modify configuration by placing additional file(s) into above mentioned directories without editing the original
* No annoying questions about changed configuration during package upgrade
* Cleaner organization
* Trivial to remove some piece (just run `rm`)
* Various different scripts will not affect each-other, not even accidentally (unless they configure inherently conflicting options)
* Scripts can be made idempotent - that means running them multiple times (regardless if previous attempts failed or not) does not break anything. I do this in all my deb packages.

Also I suggest collecting all information into variables first and then do single `cat <<EOF` to spit it out. This is much more readable.
You may want to use some loop if the user entered incorrect value.

Ben10Brasileiro, 2021-03-31:
@pretyflaco
After several days of trying and correcting the code, finally huh !!!!

@Kixunil 
* Loop - Implemented if the user dont type correctly. I just need to implement a checkout option after inserting the bridges for the user to confirm or correct.
* cat <<EOF - Not implemented due to each variable depending on IF statement, I cant see a clear way to do this with cat
* tor-hs-config-patch - Read it and https://github.com/debian-cryptoanarchy/cryptoanarchy-deb-repo-builder/blob/master/pkg_specs/tor-hs-patch-config.sps. I understood what it does, I can see the benefits clearly, but I dont know how to run your patch, could you explain the commands? Should I just git clone the repo and change the [make file](https://github.com/debian-cryptoanarchy/cryptoanarchy-deb-repo-builder/blob/master/Makefile) for just Tor config (to not install the other things)?
* Is it possible to not use microsfot.net repo for the Tor configuration?

Kixunil, 2021-04-01:
> Not implemented due to each variable depending on IF statement, I cant see a clear way to do this with cat

Ah, I see now, you're outputting 0-3 *key-value pairs*. That's not worth adding to variables I think. Perhaps loop would be more readable if you feel like increasing code quality a bit.

> but I dont know how to run your patch,

Firstly, it really has to be inside Debian package because the package hooks-into the OS to make sure it works correctly across updates of the Tor package. There are various ways to obtain the package. The simplest is just adding my repository with a few commands and then `apt install tor-hs-patch-config`. This will install that package and tor ONLY. There are no other dependencies besides `tor`. The package is platform-independent. This relies on trusting me, of course. Since it depends on Tor, you need to mask it beforehand.

If you're interested in less trust, you can inspect the package manually (just apt download it, unpack and see the contents). It's just scripts, no binaries, so it's easy to review. Or you can [build it yourself](https://github.com/debian-cryptoanarchy/cryptoanarchy-deb-repo-builder/tree/master/docs/developer#building). If you're building only that package, you don't need most of the dependencies, just debcrafter, make and mustache but you need to specify `SOURCES=tor-extras`. However, to really properly review it you would need to also review debcrafters code (few thousands LoC), so I believe inspecting the built package is easier.

> Is it possible to not use microsfot.net repo for the Tor configuration?

Yes, the package doesn't depend on .net so it wouldn't get installed anyway.

Ben10Brasileiro, 2021-04-01:
@Kixunil 
```
gpg --keyserver hkp://keyserver.ubuntu.com --recv-keys 3D9E81D3CA76CDCBE768C4B4DC6B4F8E60B8CF4C
gpg --export BC528686B50D79E339D3721CEB3E94ADBE1229CF | sudo apt-key add -
echo 'deb [signed-by=3D9E81D3CA76CDCBE768C4B4DC6B4F8E60B8CF4C] https://deb.ln-ask.me/beta buster common local desktop' | sudo tee /etc/apt/sources.list.d/cryptoanarchy.list > /dev/null
sudo apt update
sudo apt download tor-hs-patch-config (Just to read it)
sudo apt install ./tor-hs-patch-config
```
I removed Tor to be installed together with your package. Tor start first without your configuration. Then I just restart tor@default but keeps giving this errors. The same result happens with or without a torrc inside torrc.d/torrc. torrc.d and hidden-services.d are inside /etc/tor.
[image]

Kixunil, 2021-04-01:
FYI, you don't need `sudo` for `apt download` but that's non-issue here.

> I removed Tor to be installed together with your package.

You mean you removed `tor` from `Depends:` field of the package or you removed tor before installing the package?

The former is not a good idea as `tor` has to be installed before that package (but the service doesn't have to be running). It doesn't matter whether tor was already installed or is installed automatically as the dependency of `tor-hs-patch-config`.

Anyway it looks like you installed it correctly because otherwise you'd get the error during "patching ..." step.

> Also noticed this inactive during installation.

That refers to apparmor being inactive which I don't believe is reality because `aa-enabled` returned true. That looks like something is profoundly retarded in the whole crapparmor fake security system. (There were already multiple problems with it.)

You can try force reload of apparmor by `systemctl reload apparmor`. If it fails try with `sudo apparmor_parser -r /etc/apparmor.d/system_tor`.

It needs to be fixed in the package but frankly I don't know how to do it correctly without breaking existing stuff. This needs more research.

Ben10Brasileiro, 2021-04-01:
@Kixunil THANK YOU. Now it is working.
Another package non related to this was making apparmor fail (snapd, briar needed it...). I had to uninstall it, then ran this `sudo apparmor_parser -r /etc/apparmor.d/system_tor.`
Now it is all good.

Kixunil, 2021-04-01:
Oh, forgot to add two things:

* Tor has really crappy error messages (I've read the code there and that part is quite shitty), in reality it can't read the directory but this information is lost during unwinding so the output doesn't have it
* You wrote: "The same result happens with or without a torrc inside torrc.d/torrc. torrc.d and hidden-services.d are inside /etc/tor." - please do not move configuration files around like this, keep it the way it is, just add your own snippets to appropriate directories

rootzoll, 2021-04-08:
Hi everyone ... please check/review PR #2155 @nyxnor to solve this issue.