raspiblitz/raspiblitz #2056: Tor by Default

rootzoll, 2021-03-14:
So the goal is to have Tor now by default switched on. This will make the setup shorter (one option less) and also enables to use Tor already as much as possible thru the setup. Also for a future Web-GUI ist can start with offering a Tor-Webseite for setup.

The following sub tickets are prt of this bigger mission:
- https://github.com/rootzoll/raspiblitz/issues/1741
- https://github.com/rootzoll/raspiblitz/issues/1341
- https://github.com/rootzoll/raspiblitz/issues/1990
- https://github.com/rootzoll/raspiblitz/issues/2013
- https://github.com/rootzoll/raspiblitz/issues/1973
- https://github.com/rootzoll/raspiblitz/issues/1888
- https://github.com/rootzoll/raspiblitz/pull/1819
- https://github.com/rootzoll/raspiblitz/issues/1652
- https://github.com/rootzoll/raspiblitz/issues/1341

Kixunil, 2021-03-17:
What kind of web UI are you planning? I have some [nice UI](https://github.com/Kixunil/selfhost-dashboard) and working on more. Feel free to use it if you like it.

It's designed to integrate with `selfhost` packages from my repository, but should be easy enough to do it outside or just use the base selfhost packages. They are very small and platform-independent. I know of someone who already tried these things on RPi and it works without huge problems. There's a bug with dashboard not reconnecting to postgres but that's should be easy to fix and I plan to do it soon.

`selfhost` supports both Tor and clearnet and is quite clean and composable (although not as much as I want it to be) so maybe that's useful too.

rootzoll, 2021-03-17:
@Kixunil thanks for info on your WebUI ... we plan to go for a WebUI for the v1.8 release. I moven that over to the following: https://github.com/rootzoll/raspiblitz/issues/43

Ben10Brasileiro, 2021-03-18:
I think https://github.com/rootzoll/raspiblitz/issues/2054 is also a ticket for Tor by default, cause if @pretyflaco can't surpass that, he won't be able to use Blitz.
Consider tor replace the Tor packages on source.list as mentioned here https://github.com/rootzoll/raspiblitz/issues/2054#issuecomment-801091282
Explanation: 
* Another layer of protection, I think `UseBridges 1` on torrc might do it for censorship
* Adding anonimity with APT over Tor for TOR packages would be awesome if added by default.
* It is a v2 address, so a v3 attack to the tor network wont affect.

openoms, 2021-03-18:
@Ben10Brasileiro just keep in mind that v2 services will be retired starting from 15th July 2021: https://blog.torproject.org/v2-deprecation-timeline

Ben10Brasileiro, 2021-03-18:
@openoms That is a legitimate concern.
Mirrors are another option, if they are not blocked too (but the ISP can log the DNS queries, if the user dont take some actions, so maybe someday it could notice it.This is why I present the solution below (tested on desktop, but on the pi would be the same thing using the raspbian archive site with tor:// at the beggining).

I changed https with tor on the url, with the clearnet Tor project site, but this time the request will be made with tor (I think that is how it works based on it actually hitting, and the tor project using tor:// at the beggining.
To back this claim:
* https://askubuntu.com/questions/381377/apt-get-update-through-tor, 
* and this archived repo from a really long time ago https://github.com/timretout/apt-transport-tor.

I tried with `tor://` and `tor+http://` but both worked, did not see a difference.
```
deb        tor://deb.torproject.org/torproject.org      buster            main
deb-src    tor://deb.torproject.org/torproject.org      buster            main
```

Output
```
root@debian:/etc/apt# apt update
Hit:4 tor://security.debian.org/debian-security buster/updates InRelease         
Hit:9 tor://deb.debian.org/debian buster InRelease                           
Hit:10 tor://deb.torproject.org/torproject.org buster InRelease                 
Hit:12 tor://deb.debian.org/debian buster-updates InRelease
Fetched 9,511 B in 2s (4,498 B/s)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
```

rootzoll, 2021-04-08:
OK to most important issues are part if the v1.7 release ... some smaller parts are pushed to v1.7.1

nyxnor, 2021-04-13:
@openoms Found the v3 repo:
Onion list: https://onion.torproject.org/ // http://xao2lxsmia2edq2n5zxg6uahx6xox2t7bfjw6b5vdzsxi7ezmqob6qid.onion/
Tor v3 repo:
http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/

rootzoll, 2021-04-19:
OK ... this looks & tested good so far for v1.7 release. Closing issue.

closed 2021-04-19 by rootzoll