raspiblitz/raspiblitz #2062: LND stuck pending channel

jdh3, 2021-03-16:
So after letting the Raspiblitz work on autopilot for a while.  I tried to manually open a channel but set the wrong fee.  Its been about 3-4 weeks now.  The transaction failed of course.
 tx 
af784291b56651628f715cefb865c28bbd1eb6ce5564f1d8951e4787c9dc9777

I can't seem to clear it from the LND client or force it to close.

Any advice?  I've tried a lot of different things.

openoms, 2021-03-16:
See the choices: https://t.me/RaspiBlitzFAQ/234
https://bitcoin.stackexchange.com/questions/102180/can-a-channel-be-closed-while-the-funding-tx-is-still-stuck-in-the-mempool

jdh3, 2021-03-16:
Thanks for the help but I think Im past the point of trying to bump or spend the fee.  The error I see is 
Transaction invalid: some outputs that it attempts to spend have been already spent or are no longer valid.
Not sure where to go from there.  So I have and invalid transaction and my funds are still tied up with the LND wallet.

jdh3, 2021-03-16:
When I try abandon channel I get and error message as well.

openoms, 2021-03-16:
After two weeks of non-confirmation the LND wallet of yours and the peer's will forget about the channel.

It becomes a stale channel with the funds locked. AFAIK it can only be spent with the coordination of the peer and using these commands:
https://github.com/guggero/chantools/blob/master/doc/chantools_rescuefunding.md
And by the peer: https://github.com/guggero/chantools/blob/master/doc/chantools_signrescuefunding.md

How much is at stake (does it worth to work for it)?
Can you contact your peer?

openoms, 2021-03-16:
If both above are true you can install chantools with this scripts: https://github.com/openoms/lightning-node-management/blob/master/lnd.updates/bonus.chantools.sh

jdh3, 2021-03-16:
February 2nd was the the first day of the transaction so its been well over a month.  It just says pending.  Seems like there would be a simple way to rescan the wallet because the transaction failed.   Maybe Im missing something.
Thanks again for the help.  Ill keep digging.

https://www.blockchain.com/btc/tx/af784291b56651628f715cefb865c28bbd1eb6ce5564f1d8951e4787c9dc9777

openoms, 2021-03-16:
never use blockchain dot com

Blockstream.info does not have the transaction. It is likely dropped and your sats are back form the wallet you sent it from .Lnd onchain probably.
If so to make the now meaningless pending channel disappear you can use https://github.com/guggero/chantools/blob/master/doc/chantools_removechannel.md

nyxnor, 2021-04-18:
@jdh3 Was this solved? Close if positive.

jdh3, 2021-04-19:
not solved but tired of fighting with it.

closed 2021-04-19 by jdh3

nyxnor, 2021-04-19:
@jdh3 Well, this is not solved, did chantools mentioned by Openoms helped you?

little-ear, 2021-04-25:
I have a similar issue. Pending channel has no valid transaction ID for more than 14 days. I will try to find a solution with chantools. Crossing my fingers. 

I will report back.

little-ear, 2021-04-25:
Can I install and run chantools on my raspiblitz? Sorry for the rather simple question, but I found no answer to this question.

nyxnor, 2021-04-25:
@little-ear 
>How much is at stake (does it worth to work for it)?
Can you contact your peer?
If both above are true you can install chantools with this scripts: https://github.com/openoms/lightning-node-management/blob/master/lnd.updates/bonus.chantools.sh

Yes, you can install chantools.
Menu > Services > Chantools

little-ear, 2021-04-26:
1) I could not find Chantools via the Menu > Services > Channtools
2) I managed to get chantools to the raspiblitz; however I couldn't find the channel.db. It is not on '~/.lnd/data/graph/mainnet/channel.db'. I have '/home/admin/.lnd/data/chain/bitcoin/mainnet' containing *.macaroon

where do I find the (standard) channel.db folder?

nyxnor, 2021-04-26:
@little-ear `/mnt/hdd/lnd/data/graph/mainnet/channel.db`

rootzoll, 2021-04-27:
> Menu > Services > Channtools

if part of the new v1.7 release - you may need update first.
https://github.com/rootzoll/raspiblitz#downloading-the-software

little-ear, 2021-04-28:
I think my lnd channel DB is somewhat of "locked" or in read only mode for the 'bitcoin' user. 

So what happened. 
1) I updated raspiblitz to 1.7, and indeed now I was able to install chantools automatically from the 'menu' (great!)
2) Within the menu, I just pressed ESC to enter into the terminal (in Version 1.6 there was an option in the 'menu'; since 1.7 this option is gone) 
3) Once in the Terminal, you are the user 'admin' 
4) Once you enter a chantools command, you are automatically switched to the 'bitcoin' user and you are asked to reenter your command (very nice done) 
5) so I typed again: 
bitcoin@nameofpi:~ $ chantools removechannel --channeldb ~/.lnd/data/graph/mainnet/channel.db --channel 465......92:0

Output: 
2021-04-28 13:24:20.754 [INF] CHAN: chantools version v0.8.2 commit 
Error: error opening channel DB: timeout

Sorry for bothering you again. But I cant find the solution on how to unlock the channel db (e.g. https://github.com/guggero/chantools/blob/master/doc/chantools.md), or how to proceed from this error

openoms, 2021-04-28:
@little-ear first you must stop lnd
`sudo systemctl stop lnd`

and be careful not to remove any channels with funds.

chantool simply removes the channel from the database. That command does not recover funds, so make sure your funds sent  to that channel are now safe (not in a pending or confirmed transaction to a 2of2 multisig channel).

little-ear, 2021-04-29:
Okay - here is my final report. Form my point of view, this topic can be marked as solved: 

PROBLEM: 
After opening a channel with a peer, the associated transaction ID was never submitted to the bitcoin chain nor in mempool. 

VERIFICATION OF THE PROBLEM:
1) Waiting more than 14 days for the transaction ID to be submitted.
2) Thereafter, verification on blockstream.info (paste your transaction ID). My transaction ID was not found. 

Solution: 
1) Update to raspiblitz 1.7 
2) Install chantools in `raspiblitz` menu
3) Verify again (just to be sure that your transaction was not submitted in the meantime)
4) Exit the 'raspiblitz' menu (press ESC)
5) As 'admin' user run `sudo systemctl stop lnd`
6) Switch to 'bitcoin' user (if you use `chantools removechannel ...` as 'admin' user, you are switched automatically to 'bitcoin' user)
7) Run `chantools removechannel --channeldb ~/.lnd/data/graph/mainnet/channel.db --channel 465...d92:0`
where `465...d92:0` is your transaction ID
8) Exit 'bitcoin' user
9) Run `sudo systemctl start lnd`
10) Run `raspiblitz`
11) There you will have to retype your password c (just follow the instructions) 
12) I just did a reboot (not sure if needed) 
13) Re-login to start raspiblitz (type in your password c) 
14) I was happy that it worked for me. ;-) 

However, there is no guarantee that this will solve your specific problem! It just worked in my case. Do your own research! Apply at your own risk ;-)

openoms, 2021-04-29:
thank you @little-ear for summarising the steps this clearly. It is the solution for similar issues until LND will have the removechannel function itself.

little-ear, 2021-05-02:
I am really sorry for bringing this topic up again. But I might have encountered a real bug. 

As previously described my (unconfirmed and pending) channel was closed (chantools) and my sats where back (not in bracket or UTXO). 
However, I rechecked my blitz now, and noticed that my previous transaction (for opening a channel) got into mempool. It can be now found on https://blockstream.info. 
Interestingly my channel is closed - so this transaction is going nowhere I guess. 

Is it possilbe to double spend the value with raspiblitz (Thunderhub or RTL)? 
Do I miss something obvious?

openoms, 2021-05-04:
You can only double spend (using RBF - called Cancel) with the latest Electrum.
Do it before the transaction confirms!
https://openoms.gitbook.io/lightning-node-management/restorelndonchainfundsinelectrum