raspiblitz/raspiblitz #2066: Test & make sure updated apps run correctly

rootzoll, 2021-03-16:
merged #2043 .. but now make sure that all those apps run correctly.

BTCpayServer has still error:
`Mar 16 23:14:36 raspberrypi dotnet[12088]: Unhandled exception. System.InvalidOperationException: No database option was configured.` see https://github.com/rootzoll/raspiblitz/pull/2043#issuecomment-800674779

openoms, 2021-03-17:
BTCpay is fixed in: https://github.com/rootzoll/raspiblitz/pull/2067

rootzoll, 2021-03-17:
Worked in Test:
-[X] BTCPayServer
-[X] Thunderhub
-[ ] Specter
- [ ] joininbox & joinmarket
- [ ] BOS

rootzoll, 2021-03-17:
OK my first tests look good - keeping this issue open for final testing before release.

rootzoll, 2021-04-18:
OK .. apps looking good. Closing issue for v1.7 release.

closed 2021-04-18 by rootzoll