raspiblitz/raspiblitz #2069: add vanguards

saloppe73, 2021-03-17:
possible to add vanguards for tor users?

rootzoll, 2021-03-17:
For readers background:
https://blog.torproject.org/announcing-vanguards-add-onion-services

@openoms have you experience in using the vanguards add-on for Tor?

nyxnor, 2021-04-18:
Repo and torrc config
https://github.com/mikeperry-tor/vanguards#onion-service-use

nyxnor, 2021-06-25:
<nyxnor> In vanguards specs, it says that it is for onion services with a lot of traffic, but I assume that this is considering that an attacker can stumble uppon. Can I say that vanguards can server to any public onion service, regarding hidden service for personal use is overkill? Or even for hidden ones it could benefit from the protection? https://github.com/mikeperry-tor/vanguards

<mikeperry> nyxnor: vanguards is for any service that needs anonymity. most attacks against onion services work by generating a lot of connections or traffic to the onion service... so it doesn't mater if the onion service is not high traffic normally. an attacker can make as many connections to it as they want

nyxnor, 2021-06-25:
Regarding vanguards, it will be an option for tor.hidden-service.sh (internet.tor.sh).
Will need to run 2 instances for each control port (default and lnd) 9051 and 9071.
It is totally optional, installed through the menu, can start, stop, remove install at any time.