raspiblitz/raspiblitz #2070: Make it easy to move from Umbrel to RaspiBlitz

rootzoll, 2021-03-17:
More and more people asking how to move from Umrel to RaspbiBlitz. A cool feature for the future would be just to remove the umbrel sd card and inject a RaspiBlitz card. Then RaspiBlitz would when seeing a formated drive with umbrel data kick of a migration dialog.

- warn the user that only the LND wallet with channels will be moved (data of additinal apps may be deleted)
- create a RaspiBlitz Migration file from umbrel data (put unsed umbrelo data into a umbrel_old folder)
- - offer that miogration file for download (justin case)
- copy LND data to sd card (will that fit? even with a bigger lnd database?)
- format SSD
- automatically use Migration file from sd card or ask user to upload again

rootzoll, 2021-03-26:
Also while on the topic ... also take a look at myNode.

rootzoll, 2021-03-31:
As a background ... see how umbrel instructs docker apps to use the `{APP_DATA_DIR}` and maybe see later on how this can be mapped to exporting an migration file. https://github.com/getumbrel/umbrel/blob/master/apps/README.md

rootzoll, 2021-04-03:
OK the basics for documentation:

Umbrel uses `ext4` for their data drive and mounts it on `/mnt/data`:
[image]

All umbrel data on this HDD is in: `/mnt/data/umbrel` ... for the beginning we should store the following files/directories for migration:

```
/mnt/data/umbrel/bitcoin/wallet.dat
/mnt/data/umbrel/lnd
```

and also put a copy of the following data into a special `migration_backup` folder:
```
/mnt/data/umbrel/info.json
/mnt/data/umbrel/db
/mnt/data/umbrel/apps
/mnt/data/umbrel/app-data
```

rootzoll, 2021-04-03:
myNode uses `ext4` for their data drive and mounts it on `/mnt/hdd`:
[image]

All myNode data on this HDD is in: `/mnt/hdd/mynode` ... for the beginning we should process the following files/directories for migration:

```
/mnt/hdd/mynode/bitcoin
/mnt/hdd/mynode/lnd
```

the rest of the HDD could stay in the `mynode` folder and renamed into `migration_backup`

rootzoll, 2021-04-05:
OK merged a basic implementation ... now syncing some umbrel & mynode nodes to make final testing and improvements on.

rootzoll, 2021-04-08:
Please be careful testing these on v1.7RC2 with funded notes ... I was not able to make proper test runs. So final testing and improvements are happening on these features after 1.7RC2 release.

openoms, 2021-04-10:
used a disk with synced Bitcoin Core in the folder `bitcoin` and came to this:
```
![image](https://user-images.githubusercontent.com/43343391/114273159-fc6ba780-9a10-11eb-9da7-18284e947e16.png)

Starting the main menu ...
**************************************************
MIGRATION FROM MYNODE TO RASPIBLITZ
**************************************************
- started ...
/dev/fd/63: line 1: IMPORTANT: command not found
- free space of data disk is low ... deleting 'backup_migration'
rm: cannot remove '/mnt/hdd/backup_migration': No such file or directory
******************************************************
OK MIGRATION --> will now reboot and update/recover
******************************************************
admin@raspberrypi:~ $ reboot
User pi is logged in on tty1.
Please retry operation after closing inhibitors and logging out other users.
Alternatively, ignore inhibitors and users with 'systemctl reboot -i'.
admin@raspberrypi:~ $ sudo shutdown now -r
Connection to 192.168.3.161 closed by remote host.
Connection to 192.168.3.161 closed.
```
This was going in a loop of offering migration and restart until I srlected `Ignore`.

Not bad for a custom state, just a bit more explanation needed to say to press ignore if not coming from mynode.
Also the restart is not automatic.

[image]
Afterwards the blockchain-data recovery went as expected.

rootzoll, 2021-04-16:
Ok I tested now with a Umbrel that had a channel open ... and migration worked - was able to continue that channel with raspiBlitz without closing. Testing & improving on myNode next.

rootzoll, 2021-04-16:
> used a disk with synced Bitcoin Core in the folder bitcoin and came to this:

@openoms your right - the detection method yor myNode is not optimal ... I will change thanks.

rootzoll, 2021-04-16:
OK the migration of myNode is now changed & implemented and ran sucessfull with one open channel. There was the need to refactor the change of password C that now will be aked on final setup after migration to change from random walletpassword of myNode to user set password c of the user for RaspiBlitz.

nyxnor, 2021-04-18:
@pretyflaco You got mynode detected warning. When RC3 comes out, please test it.

rootzoll, 2021-04-19:
Make final test with latest myNode release - see if still works: https://twitter.com/mynodebtc/status/1384270459684876296?s=20

rootzoll, 2021-04-23:
OK myNode migration tested good. 

Umbrel just released v0.3.9 with some changes on booting from SSD. Will run last migration test tomorrow to see if this constains some changes affecting the migration. https://twitter.com/getumbrel/status/1385593991559716874?s=20

rootzoll, 2021-04-24:
Looks like Ineed to adapt one thing for umbrel mibration. With the new version the changed to a fixed/constant LND wallet password called `moneyprintergobrrr`- see https://github.com/getumbrel/umbrel-manager/blob/7e9f458fb9616fa60a5160368fcd14a1d0cfb639/utils/const.js#L29

rootzoll, 2021-04-24:
OK added the needed adaption for umbrel 0.3.9 ... tested good, but now need to test if its still working with older umbrel version before release.

rootzoll, 2021-04-24:
tested again migration with old Umbrel version and its also still working. So all looks good - RaspiBlitz should be able to handle Umbrel v0.3.8 & v0.3.9 for migration. Will be part of v1.7 release. Closing issue.

closed 2021-04-24 by rootzoll

lapistano, 2021-07-16:
Hey Guys,
just one word ... awesome !!! 

is there an update to the script for umbrel 0.3.14 und raspibliz 1.7.0?

nyxnor, 2021-07-16:
@lapistano 
Not yet especifically for the 0.3.14, last version tested by the devs was 0.3.8, but people were still able to migrate with Umb version 0.3.13 https://github.com/rootzoll/raspiblitz/issues/2369#issuecomment-869057327, so you should be fine with .14.
Keep a backup on a separate drive for safety.

bitpaint, 2022-04-07:
Hey guys, the link here is going to nowhere: 

https://github.com/rootzoll/raspiblitz#make-a-raspiblitz-out-of-your-umbrel-or-mynode

[image]

openoms, 2022-04-07:
The link is to be corrected to https://github.com/rootzoll/raspiblitz#make-a-raspiblitz-out-of-your-umbrel-citadel-or-mynode

nyxnor, 2022-04-07:
for whomever is PRing, run `./toc.sh README.md` to recreate the table of contents every time the markdowns are updated, there may be more outdated links.

The last update was on feb 9 and merged feb 10 https://github.com/rootzoll/raspiblitz/pull/2943/commits
but the readme was updated on feb 20 https://github.com/rootzoll/raspiblitz/commit/42498b0bec07917af3ceaccfd5a533322a6267ca

rootzoll, 2022-04-25:
Updated link will be part of v1.8. Also marked TOCs to be refreshed for v1.8