raspiblitz/raspiblitz #2073: [Feature request] [Privacy] run a parallel second LND node for private donations and to connect to Sphinx chat

openoms, 2021-03-18:
Aimed to be an advanced (CLI only) privacy feature for Tor only nodes.
Not aimed for routing, only needs one private (ideally coinjoined) and appropriately sized (>100k sats) UTXO to open one channel.

To be used for Sphinx chat and to receive donations a channel needs to be published: 
  * can be opened to https://1ml.com/node/023d70f2f76d283c6c4e58109ee3a2816eb9d8feb40b23d62469060a2b2867b77f/channels as in  https://github.com/stakwork/sphinx-relay#if-you-just-installed-your-lnd
  * keysend only works if the channel is public
  * receiving the donations the funding UTXO would need to be exposed in the route hint if the channel is private
  
By generating the connection string the node can be used for BTCpay private donations.
* incoming capacity can be created by paying out to another node or mobile wallet.

A separate channel could be private for the most private payments  - in that case the existence of the Lightning channel is not advertised publicly.

Balance of Satoshis can be configured to interact with multiple nodes:
* https://github.com/alexbosworth/balanceofsatoshis#saved-nodes
* using `bos open` can reduce the channel funding to a single transaction from any PSBT/rawTransaction capable wallet (JoinMarket, Specter, Electrum)

- [ ] install script with 
- [ ] separate user and data directory (symlinked to app-data)
- [ ] specify separate ports for RPC, REST and GRPC
- [ ] Sphinx chat connection
- [ ] BTCpay string
- [ ] Zeus connection (lndconnect)

lnd.conf (see: https://openoms.gitbook.io/lightning-node-management/lnd.conf)
```
[Application Options]
# alias=ALIAS # up to 32 UTF-8 characters
# color=COLOR # choose from: https://www.color-hex.com/
listen=0.0.0.0:9736
rpclisten=0.0.0.0:10010
restlisten=0.0.0.0:8081
accept-keysend=true
nat=false
debuglevel=debug
gc-canceled-invoices-on-startup=true 
gc-canceled-invoices-on-the-fly=true 
ignore-historical-gossip-filters=1 
sync-freelist=true
stagger-initial-reconnect=true
tlsautorefresh=1
tlsdisableautofill=1

[Bitcoin]
bitcoin.active=1
bitcoin.node=bitcoind

[Bitcoind]
bitcoind.rpcuser=RPC_USER
bitcoind.rpcpass=RPC_PASSWORD
bitcoind.rpchost=127.0.0.1
bitcoind.zmqpubrawblock=tcp://*:28332
bitcoind.zmqpubrawtx=tcp://*:28333
bitcoind.estimatemode=ECONOMICAL

[Wtclient]
wtclient.active=1

[Tor]
tor.password=PASSWORD_B
```

openoms, 2021-03-22:
Useful notes from: https://github.com/lightningnetwork/lnd/blob/master/docs/INSTALL.md#using-bitcoind-or-litecoind
NOTE:

- The auth parameters rpcuser and rpcpass parameters can typically be determined by lnd for a bitcoind instance running under the same user, including when using cookie auth. In this case, you can exclude them from the lnd options entirely.
- If you DO choose to explicitly pass the auth parameters in your lnd.conf or command line options for lnd (bitcoind.rpcuser and bitcoind.rpcpass as shown in example command above), you must also specify the bitcoind.zmqpubrawblock and bitcoind.zmqpubrawtx options. Otherwise, lnd will attempt to get the configuration from your bitcoin.conf.
- You must ensure the same addresses are used for the bitcoind.zmqpubrawblock and bitcoind.zmqpubrawtx options passed to lnd as for the zmqpubrawblock and zmqpubrawtx passed in the bitcoind options respectively.
- When running lnd and bitcoind on the same Windows machine, ensure you use 127.0.0.1, not localhost, for all configuration options that require a TCP/IP host address. If you use "localhost" as the host name, you may see extremely slow inter-process-communication between lnd and the bitcoind backend. If lnd is experiencing this issue, you'll see "Waiting for chain backend to finish sync, start_height=XXXXXX" as the last entry in the console or log output, and lnd will appear to hang. Normal lnd output will quickly show multiple messages like this as lnd consumes blocks from bitcoind.
- Don't connect more than two or three instances of lnd to bitcoind. With the default bitcoind settings, having more than one instance of lnd, or lnd plus any application that consumes the RPC could cause lnd to miss crucial updates from the backend.
- The default fee estimate mode in bitcoind is CONSERVATIVE. You can set bitcoind.estimatemode=ECONOMICAL to change it into ECONOMICAL. Futhermore, if you start bitcoind in regtest, this configuration won't take any effect.

Edited the lnd.conf above accordingly. (No need of rpcuser + password)

openoms, 2021-03-22:
The LND installation part can be done with:
https://github.com/openoms/joininbox/blob/master/scripts/install.lnd.sh (part of the latest v0.3.3 JoininBox release)
tested on the RaspiBlitz (`admin` or `joinmarket` user) and standalone JoininBox as well.
Installs an LND node alongside the existing one on the RaspiBlitz.

To download separately:
```
# download
wget https://raw.githubusercontent.com/openoms/joininbox/master/scripts/install.lnd.sh
# inspect 
cat install.lnd.sh
# install: 'install.lnd.sh [on] [NODENUMBER(default=1)]'
bash install.lnd.sh on 1
```

rootzoll, 2021-03-22:
just one thought .. wouldnt the accounts in LND Oliver is working on simulate a "second LND node" for use with sphinx?

openoms, 2021-03-22:
Accounts would not get new pubkeys (like they are not in LNBits or lndhub/Bluewallet).

Generating new pubkeys should be possible with the HTLCinterceptor API, but did not see it working yet and then not sure if those could be connected to Sphinx.

This is a solution I need and works now, but would be happy if it is only temporary.

openoms, 2021-03-23:
New version of the script with most of the features added: https://github.com/openoms/joininbox/blob/dev-lnd/scripts/install.lnd.sh 
Only tested the installation and adding to BoS so far. Just waiting for a channel to confirm and my private node will be functional!

```bash
bash  install.lnd.sh -h
# script to install LND
# the default version is: v0.12.1-beta
# install.lnd.sh [on <nodenumber>|off <nodenumber> <purge>
# install.lnd.sh [add-node-to-bos <nodenumber>]
# 
# to use a node with Sphinx-relay
# install.lnd.sh [write-sphinx-environment <nodenumber>]
# 
# to export macaroons & tls.cert:
# install.lnd.sh [exportmenu|hexstring|scp|http|btcpay <nodenumber>]
```

```
# download
wget https://raw.githubusercontent.com/openoms/joininbox/dev-lnd/scripts/install.lnd.sh
# inspect 
cat install.lnd.sh
# install: 'install.lnd.sh [on] [NODENUMBER(default=1)]'
bash install.lnd.sh on
```

openoms, 2021-03-24:
Relevant tweet thread: https://twitter.com/openoms/status/1374772296934699015

openoms, 2021-03-27:
Step-by-step instructions:
* install the lnd1 node
```
# download the script (force overwrite)
wget -O install.lnd.sh https://raw.githubusercontent.com/openoms/joininbox/dev-lnd/scripts/install.lnd.sh
# inspect 
cat install.lnd.sh
# install: 'install.lnd.sh [on] [NODENUMBER(default=1)]'
bash install.lnd.sh on
```

* Connect to bos / thunderhub / sphinx:
```
bash install.lnd.sh menu
```
[image]

Bos, Thunderhub and the Sphinx relay needs to be installed beforehand. 
For Sphinx to work the best use the new script to install the latest version with fixed permissions:
https://github.com/rootzoll/raspiblitz/issues/2089#issuecomment-805979006

* for bos it adds an alias accessible as `bos1`, the existing config is untouched
* for THub there will be a second node to log into the Thub webpage with the same password (PasswordB)
* Sphinx can only be connected to one LND at a time.  
The old database (if exists) is backed up and a new database is created for lnd1. 
Can be paired to the mobile app from the SPHINX menu in the raspiblitz main menu.

Can follow this: https://github.com/stakwork/sphinx-relay#if-you-just-installed-your-lnd
* the CLI of the new lnd1 node can be accessed with `lncli1`
* can also use Thunderhub to fund and open a channel if added already
* open a channel with min 100k sats to the Sphinx node (as in the readme above): https://1ml.com/node/023d70f2f76d283c6c4e58109ee3a2816eb9d8feb40b23d62469060a2b2867b77f 
to be well connected and benefit from no (or minimal) hops (free LN transactions) when using Sphinx chat

If you can join the RaspiBlitz tribe on Sphinx we know it all worked.
[image]

fourninefour, 2021-03-28:
Hi, thanks for the step-by-step instructions! I was able to follow it to get to the desired result of having a sphinx-relay running on a separate lnd instance. I had some issues since due to a force-closed channel, but I would assume that this does not relate to this topic directly.

I did not precisely keep track of all the exact steps I took and the issues I ran into, but here are some of the things I encountered:
- Running `bash install.lnd.sh on` returned `FAIL: unknown ''`, but it still seemed to work and I had `lnd1` running afterwards and `lncli1` working. The `FAIL: unknown ''` error was raised again in (I believe) the sphinx-relay install.
- Running THUB in the MENU multiple times added the `lnd1` instance multiple times. Maybe this was intended, but an alternative could be to check if it already exists and replace it instead, to only have a single `lnd1` as a result.
- BOS initially complained that `/home/bos/.bos doesn't exist`, even though it was installed on my raspiblitz. It worked after re-installing and updating BoS.
- I had trouble connecting to Sphinx, so I ended up re-installing it and restarting my raspiblitz many times. I am not sure what the issues were, unfortunately. But in the end it worked.

openoms, 2021-03-28:
thansk for the points.
1. solved - redownload the script
2. can improve, for now edit the thub conf manually:
`sudo nano /mnt/hdd/app-data/thunderhub/thubConfig.yaml`
3. couldn't reproduce this
4. solved

nyxnor, 2021-05-09:
Openoms, can this be closed? 
Also, could you update this link: https://keybase.pub/oms/slides/Running_a_Lightning_Node_Privately.pdf
Not Found.

openoms, 2021-09-27:
Closing this for inactivity. Did not go forward with further integration.

@nyxnor try downloading from https://keybase.pub/oms/slides

alaznem, 2021-11-26:
> https://raw.githubusercontent.com/openoms/joininbox/master/scripts/install.lnd.sh

@openoms : This url shows a 404 error.

:question: Could you explain what needs to be done to run a parallel second LND node on raspiblitz at the moment (still for sphinx only)?

rootzoll, 2021-11-26:
@alaznem we dropped the concept of running multiple LND mainnet nodes.

Instead with v1.7.1 its now possible to run LND & c-lightning in parallel ... that way you also have two lightning nodes and you can use LND for sphinx/personal-use while c-lightning for your anon routing node.

openoms, 2021-11-27:
@alaznem the script is accessible at https://raw.githubusercontent.com/openoms/joininbox/v0.6.2/scripts/install.lnd.sh.
It was removed from joininbox as not planning to support due to too much added complication with backups, recovery and migrations. Agree with rootzoll's comment above.