raspiblitz/raspiblitz #2074: Add experimentell docker support for additional apps

rootzoll, 2021-03-18:
So additional apps asked if its possible to add docker - I did some testing and it should be possible to have experimental docker support for v1.7 release.

rootzoll, 2021-03-23:
OK `home/admin/config.scripts/blitz.docker.sh`will be part of v1.7RC1 .. additional apps can use this script to make sure that docker is available.

PLEASE NOTE: This is just the first adding of docker ... more `structure` how to config additional docker apps may come in the future to gain compatibility with umbrel or mynode docker apps. Also starting with v1.7RC1 we will run on arm-64bit RaspiOS.

fusion44, 2021-04-05:
The install script references `/home/admin/assets/get-docker.sh` which doesn't seem to be available in current dev branch.

openoms, 2021-04-12:
Still in v1.7RC2:
```
admin@raspberrypi:~ $ config.scripts/blitz.docker.sh on
### 1) INSTALL docker ###
chmod: cannot access '/home/admin/assets/get-docker.sh': No such file or directory
sudo: /home/admin/assets/get-docker.sh: command not found
usermod: group 'docker' does not exist
Failed to start docker.service: Unit docker.service not found.
### 2) INSTALL docker-compose ###
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting docker-compose
```

openoms, 2021-04-14:
Testing docker in: https://github.com/openoms/lightning-benchmark/blob/master/install.aarch64.md

Installation on the 64bit image is straightforward 
* note the symlink of /var/lib/docker from the SSD)
* used a RaspiBlitz v1.7 base image (based on https://www.raspberrypi.org/forums/viewtopic.php?p=1668160)

## Install Docker
```
# dependencies
sudo apt-get update
sudo apt-get install \
    apt-transport-https \
    ca-certificates \
    curl \
    gnupg \
    lsb-release

# add the docker repo
echo \
  "deb [arch=arm64 signed-by=/usr/share/keyrings/docker-archive-keyring.gpg] https://download.docker.com/linux/debian \
  $(lsb_release -cs) stable" | sudo tee /etc/apt/sources.list.d/docker.list > /dev/null

# install docker
sudo apt-get install docker-ce docker-ce-cli containerd.io

# add the default user to the docker group
sudo usermod -aG docker admin

# check (after relogin works without sudo)
sudo docker run hello-world
# Hello from Docker!
```

## Install docker-compose
```
sudo pip3 install docker-compose

# add bash completion  https://docs.docker.com/compose/completion/
sudo curl \
    -L https://raw.githubusercontent.com/docker/compose/1.29.0/contrib/completion/bash/docker-compose \
    -o /etc/bash_completion.d/docker-compose

# check
docker-compose --version
# docker-compose version 1.29.0, build unknown
```

## Symlink working directory to the SSD
```
sudo systemctl stop docker
sudo systemctl stop docker.socket
sudo mv /var/lib/docker /mnt/hdd/
sudo ln -s  /mnt/hdd/docker /var/lib/docker
sudo systemctl start docker
sudo systemctl start docker.socket
```

rootzoll, 2021-04-17:
OK sorry i messed stuff up with the docker script ... lets push this to v1.7.1

nyxnor, 2021-05-06:
Hi, little curiosity. The reason to Blitz not been using docker is related to security or for educational purposes?

rootzoll, 2021-05-16:
@nyxnor a bit of both. Basically the design goal is too keep dependencies on other frameworks to run basic bitcoin/lightning setup as minimal as possible to reduce attack vectors. Docker is OK for additional apps but there should always be an option that you dont need docker to run the basics.

Also it should be easy for people to have direct access to their bitcoin/lightning from the command line and not abstracting that away. And personally I find it easier to verify the setup of a system if it downloads and checks PGP signatures directly instead of pulling docker images or having to dig thru complex docker-compose files to understand whats going it. It just feels more direct and baremetal.

xanoni, 2021-08-02:
Here is a tutorial that explains the process on 32 Bit Raspbian. I followed it and can confirm it's working.

https://withblue.ink/2020/06/24/docker-and-docker-compose-on-raspberry-pi-os.html

DocBrown101, 2022-02-28:
Are there any plans for this Docker integration to be compatible with the Umbrel Docker App Framework, at least in large parts?
https://github.com/getumbrel/umbrel/blob/master/apps/README.md

This would greatly simplify the establishment of an app ecosystem.

rootzoll, 2022-03-08:
@DocBrown101 The idea is/was there ... but too much to do first. And basically since umbrel pivoted to be a "home server" instead of being a "Dedicated Bitcoin/Lightning node" not sure if it still make sense. At least its not the goal of RaspiBlitz to offer all the same apps like umbrel anyhow anymore.

Maybe it can be benefit for resiliance of the lanscape that not all node projects depend on the same duckerhub for their app installs. Maybe going towards a debian packaging in the long run is the better way for RaspiBlitz - we will see once we get there.

But to make adding your Bitcoin/Lightning app to RaspiBlitz a bit easier there is now a template install-script developer can use and tries to explain the most important steps: https://github.com/rootzoll/raspiblitz/blob/v1.7/home.admin/config.scripts/bonus.template.sh

closed 2024-08-06 by rootzoll