raspiblitz/raspiblitz #2075: Retest HDMI switch

rootzoll, 2021-03-18:
With ne new 64-bit bas OS the LCD driver changes ... check if the HDMI switches are still working.

- the switch with the file on the sd card on setup
- the switch with the script: `/home/admin/config.scripts/blitz.lcd.sh hdmi on`

Maybe even consider autodetection with `lcdExists=$(sudo ls /dev/fb1 2>/dev/null | grep -c "/dev/fb1")` on initial setup

rootzoll, 2021-03-27:
Also putting `echo "hdmi_safe=1" >> /boot/config.txt` in the build sd card script by default was not working (LCD stayed black) -- but it can make sense when the RaspiBlitz is set into HDMI mode and removed when switched back to LCD. This should be added to the switch script for RC2.

rootzoll, 2021-04-05:
@openoms I am doing heavy refactoring on the LCD/HDMI/headless mode part of the Blitz. I would like to remove some complexity and remove the LCD drivers for alternative platforms right now. Those can work with HDMI or headless. The LCD and especially the touchscreen later should be just for our focus image the RaspiOS 64-bit for now. Would that be OK from your side?

openoms, 2021-04-06:
Sure, RPis should use the 64bit image and the rest is not compatible with this driver anyway.

rootzoll, 2021-04-07:
OK I merged the display refactor. The HDMI switch is now to be tested for release.

rootzoll, 2021-04-13:
OK tested adding a file called `hdmi` to the boot sd card and the RaspiBlitz is the switching from LCD to HDMI.
Test OK .. will be part of v1.7 RC3/final release. Closing issue.

closed 2021-04-13 by rootzoll