raspiblitz/raspiblitz #2082: Screen remains blank

josephtyler, 2021-03-21:
I ordered the screen that was linked to in the README file. I set Raspiblitz up a few weeks ago and the screen just came in today, maybe there were some missing installation steps because I didn't have the screen connected during setup?

I turned the screen on in the raspiblitz config, and after a reboot it remains blank.

Sadly, the calibrate script does nothing.

The only errors I see when turning the screen on is that `apt-get update` seems to fail, but it seems that all of the apt packages are still installed in the following lines.

Any help you can offer would be greatly appreciated.

openoms, 2021-03-21:
Reminds me of this issue: https://github.com/rootzoll/raspiblitz/issues/1992#issuecomment-790586335
you might something there which helps.

nyxnor, 2021-04-18:
@josephtyler Was this solved? Pls close if so.

nyxnor, 2021-05-17:
@rootzoll Please close this

closed 2021-05-18 by rootzoll