raspiblitz/raspiblitz #2083: [Feature request] [Onboarding] Some kind of hint to the official RaspiBlitz documentation in the setup phase

mrsellars, 2021-03-21:
When I set up my RaspiBlitz as a new user it would be great to get a hint in the beginning, for example, a link, to the official RaspiBlitz documentation or RaspiBlitz wiki where I can find more information when I get stuck in the setup.

rootzoll, 2021-03-22:
Yeah this could be part of the UI rework ... targeting for v1.8

rootzoll, 2022-05-09:
@mrsellars once the v1.8.0 with the new WebUI is out - feel free to share your ideas in more detail where to add more on-boarding information. Pushing this to v1.8.1 for now.

rootzoll, 2024-04-23:
@cstenglein now that we have the new documentation, shoudl we add a link to it during the WebUI setup?

cstenglein, 2024-04-23:
Sure => Ticket

cstenglein, 2024-10-05:
Will be added with https://github.com/raspiblitz/raspiblitz-web/pull/782

closed 2024-11-07 by rootzoll