raspiblitz/raspiblitz #2085: Watchtower client setup

Svobodan, 2021-03-21:
I am trying to set up a watchtower client and connect to a watchtower following this manual: https://github.com/lightningnetwork/lnd/blob/master/docs/watchtower.md

When I get to turning on the client I get this:
```
$ sudo lnd --wtclient.active
loadConfig: either bitcoin.active or litecoin.active must be set to 1 (true)
```
I am worried I will mess something up if I fiddle with the bitcoin.active when I do not uderstand it. How should I go about this?

openoms, 2021-03-21:
Use this guide: https://github.com/openoms/lightning-node-management/blob/master/watchtower.md

Svobodan, 2021-03-22:
Thank! I have got it up and running.

marxenegls, 2021-09-15:
yeah thanks openoms ; ) me too searched and found this 'watchtower' thread, now i almost have watchtower set up , if only the guide link was as willing : ( we'll decypher the url and get there on foot

openoms, 2021-09-15:
The directory structure has changed. 
New link: 
https://github.com/openoms/lightning-node-management/blob/master/advanced-tools/watchtower.md
or for a prettier format (and Spanish option) :
https://www.lightningnode.info/advanced-tools/watchtower