raspiblitz/raspiblitz #2089: Update Sphinx-Relay 2.0.3

rootzoll, 2021-03-22:
Make sure for the Sphix-Relay to be up-to-date and work with the latest releases of the Sphix App.
https://github.com/stakwork/sphinx-relay/releases

rootzoll, 2021-03-22:
Opened the branch https://github.com/rootzoll/raspiblitz/tree/2089-sphinx to work on all Sphinx releated issued. 

The connection test within the local network works fine ... but when using IP2Tor I get a "Cannot connect Server" error on pairing while even telnet is giving me a response on that ip/port.

rootzoll, 2021-03-22:
OK the connection problem got fixed. Improved the connection test & also checking now if already on app connected with the connection string.

rootzoll, 2021-03-22:
merged the branch ... so this issue is ready for testing

rootzoll, 2021-03-23:
@we-do-it-lu I am missing the context of your problem in this thread. Basically the Sphinx-Relay and the App version need to be up-to-date to work together. An RaspiBlitz under UPDATE there is the possibility to update the Sphinx-Relay while keeping your data. If you have problems please wait for the v1.7 release .. an early ReleaseCadidate v1.7RC1 will become available for the weekend - but its not advised to use that with your production node because its a still early version for testing and bug finding.

we-do-it-lu, 2021-03-24:
@rootzoll you perfectly answered my interrogations. thank you!

openoms, 2021-03-24:
@rootzoll just testing this. The install got stuck on this `sudo` I think: https://github.com/rootzoll/raspiblitz/blob/47f3455704b14125213c82bf58ecb4ce6aea1faa/home.admin/config.scripts/bonus.sphinxrelay.sh#L306

openoms, 2021-03-24:
Successful tests with the latest Sphinx-relay and a dedicated second, parallel node https://github.com/rootzoll/raspiblitz/issues/2073#issuecomment-805260911

Made a Tribe as well:
[image]

Connection code:
`YFsuxCApuU-TBIRWeWY_Gl52IOx2clJKZYLxDn78Z9RgcWNx9zUube7mkK4IyXQjLril-tgKZs3wOxDlWqpiyxp8jTbt`

I am:
openoms
`03acad130810892cd39fb7a1f8d9e5090337f7f3e2e3d8ba09b3370c32db284447`

openoms, 2021-03-24:
anyone in this Sphinx tribe? 
who is the lewildbeast?
I only see it now that I made a competing one sorry, but anyway I cannot join this (no route).
[image]

we-do-it-lu, 2021-03-24:
I get "There was an error. Please try again later." 
Blitz v1.6.3
Latest version of SPHINX on iPhone.

we-do-it-lu, 2021-03-24:
You can try to join mine your end if you like: 
sphinx.chat://?action=tribe&uuid=YFpmJB-parnCVOvTaNmwg5LKWXpGuU_6ISrPpf2KX1Mjrli7K272Hk7txpODKt__DrNPC5eKvcERRU4vKOB2lNK-5TZA&host=tribes.sphinx.chat
[image]

openoms, 2021-03-24:
> I get "There was an error. Please try again later." 
> Blitz v1.6.3
> Latest version of SPHINX on iPhone.

what are you doing then and where do you see the message? Try with the latest sphinx-relay install script (coming in the new release soon).

we-do-it-lu, 2021-03-24:
The message "There was an error. Please try again later." I am getting at the moment is a modal pop-up on SPHINX App on my iPhone when I scan your Tribe QR code.
I apologise for most likely having missed part of the story. 
I don't know how to "Try with the latest sphinx-relay install script"; script to run from the command line?

openoms, 2021-03-24:
@we-do-it-lu 
```
# download
wget https://raw.githubusercontent.com/rootzoll/raspiblitz/dev/home.admin/config.scripts/bonus.sphinxrelay.sh
# have a look
cat bonus.sphinxrelay.sh
# uninstall old version (keep data)
bash bonus.sphinxrelay.sh off
# install 
bash bonus.sphinxrelay.sh on
```

we-do-it-lu, 2021-03-24:
@openoms thank you, spot on, I managed to join your Tribe. No further comments.

openoms, 2021-04-16:
@rootzoll 
the Sphinx relay is now supported to connect over Tor as well so IP2Tor for it is redundant.
Needs Orbot on Android, but seems native on iOS:
https://github.com/stakwork/sphinx-relay/wiki/Home-node-FAQ#apps

rootzoll, 2021-04-16:
I opened a new issue for v1.7.1 to optimize for Tor connection and dropping IP2Tor long term for SphinxApp - see:
https://github.com/rootzoll/raspiblitz/issues/2182#issue-859681846

rootzoll, 2021-04-19:
All tests ran OK .. closing for v1.7 release.

closed 2021-04-19 by rootzoll