raspiblitz/raspiblitz #2090: CONNECT menu to collect how to connect wallets and services

openoms, 2021-03-22:
Would help with answering the questions of how I connect XX app or wallet to the node like.

By default should show connection to bitcoind and LND:

- [ ] Bitcoind RPC for Specter Desktop and a remote Joininbox / JoinMarket
https://d11n.net/connect-specter-desktor-with-raspiblitz.html
https://github.com/openoms/joininbox/blob/master/prepare_remote_node.md
- [ ] Fully Noded (from the Mobile menu)
- [ ] Bitcoind P2P - BIsq connects to v2 Tor service
https://twitter.com/openoms/status/1355885287881240578 
- [ ] LND rest and GRPC with the text with
  - [ ] TLS cert (from LNDCREDS)
  - [ ] macaroons (from LNDCREDS)
  - [ ]  hidden services
  - [ ] LNDconnect as now in the MOBILE menu fro Zeus / Zap
  
Then could change dynamically depending on what is installed:
- [ ] LND connection string for BTCpayserver (from LNDCREDS)
- [ ] Electrum Server from the ELECTRS menu
- [ ]  can consider showing the various ports to open different services
- [ ] Hidden services for the various services

rootzoll, 2021-03-23:
Just quick note: There is already a "CONNECT" menu entry for connectig to another Lightning peer. We could change that to "PEERING" ... or will that get a sub entry in the new "CONNECT" menu?

openoms, 2021-03-24:
I wonder if we could declutter the Main menu significantly.
Currently with only LN related services installed I can only see 33% of the options:
[image]

I am thinking to breakdown to a couple of submenus:
```
INFO - for the status screen
LIGHTNING - submenu for LN wallet related actions
    FUNDING
    CONNECT - can be called PEERING
    CHANNEL
    SEND
    RECEIVE
    CLOSEALL
    CASHOUT
    lnbalance
    lnchannels
    lnfwdreport
SERVICES - all the installed app info-s including:
    TOR
ADD/REMOVE - instead of the current SERVICES menu to install apps 
SETTINGS - kept as it is with the toggle options
SUBSCRIBE
CONNECT - some extra mentioned above and including:
    MOBILE
    LNDCREDS
NAME
REPAIR
UPDATE - as it is (could adjust the menu height automatically to not hide options)
REBOOT
OFF
X - can be configured to be the Cancel button instead (they do similar)
```

What do you think? it is not too much work to refactor the menu and I can do it.
Need to create 3 submenu files for LIGHTNING, SERVICES and CONNECT.

keblek, 2021-03-27:
is if possible to view this as a grid rather than a list? could have the columns be categories (LN, security, utilities, etc)

openoms, 2021-03-28:
@keblek, no the dialog system is pretty limited. A tradeoff for being extremely lightweight and compatible: https://bash.cyberciti.biz/guide/Bash_display_dialog_boxes

openoms, 2021-04-01:
#2119 is ready to be merged, but of course would benefit from more testing.
There are a lot of new menu options as described in the first comment and the commits.

One more thing I'd like to add is the watchtower info and connection to the CONNECT menu.
Should be easy to do by following: https://github.com/openoms/lightning-node-management/blob/master/watchtower.md

rootzoll, 2021-04-08:
PR merged ... putting on final testing for v1.7RC2 :)

rootzoll, 2021-04-19:
OK ... the connect menu is really nice and tested well :) Looks ready for v1.7. Closing issue for release.

closed 2021-04-19 by rootzoll