raspiblitz/raspiblitz #2101: stuck at Lightning progress: "prepare to sync"

bruceleeroy18, 2021-03-23:
I recently rebooted my raspiblitz and have been unable to get back to the LND to load. 

https://termbin.com/wjmt

https://termbin.com/4qze

Upon running LND I get the following error message. 
loadConfig: either bitcoin.active or litecoin.active must be set to 1 (true)

openoms, 2021-03-24:
Looks weird. Porbably a hardware (disk) error causing corruption of the files.
If you don't have funds in LND can try to reset the wallet from REPAIR.

If there a re funds make backups (RESCU and SCB), reflash the SD and import the RESCUE backup if not picked up automatically during recovery.

bruceleeroy18, 2021-03-24:
There are funds. I will make rescue and scb, and reflash. But if it is a hardware error do I need to get another hard drive to do this?

openoms, 2021-03-24:
you might. Make sure your USB adapter is one of the recommended ones.

bruceleeroy18, 2021-03-24:
Same error is happening. stuck at prepare sync and all lncli commands return [lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

openoms, 2021-03-24:
try reserring the TLS cert and macaroons isn LNDCREDS.
Try an LND reset and restore from the latest backup in REPAIR
if none works, you can force close your channels with the SCB, see the FAQ for the details

bruceleeroy18, 2021-03-24:
"failed to generate new LND cert"

/bin/cp: cannot stat '/mnt/hdd/lnd/data/chain/bitcoin/mainnet/admin.macaroon': No such file or directory
/bin/cp: cannot stat '/mnt/hdd/lnd/data/chain/bitcoin/mainnet/invoice.macaroon': No such file or directory
/bin/cp: cannot stat '/mnt/hdd/lnd/data/chain/bitcoin/mainnet/readonly.macaroon': No such file or directory

bruceleeroy18, 2021-03-24:
LND reset and restore didn't work.

openoms, 2021-03-24:
share some error messages if you need more help

bruceleeroy18, 2021-03-24:
okay, i could use more help. what is best way to share the error messages you are thinking of?

bruceleeroy18, 2021-03-24:
I was going to buy a new SSD, reflash raspiblitz, and try to restore from seed words and old SCB. Is that the recommended way to force close channels you mentioned?

bruceleeroy18, 2021-03-24:
lnd.credentials.sh reset
## Resetting Macaroons
# all your macaroons get deleted and recreated
rm: cannot remove '/home/bitcoin/.lnd/data/chain/bitcoin/mainnet/macaroons.db': No such file or directory
## Resetting TLS
# tls cert gets deleted and recreated
# checking if LND is running
# making sure services are not running
# keep old tls data as backup
mv: cannot stat '/mnt/hdd/lnd/tls.cert': No such file or directory
mv: cannot stat '/mnt/hdd/lnd/tls.key': No such file or directory
# start to create new generate new TLSCert
# wait until generated

openoms, 2021-03-24:
> I was going to buy a new SSD, reflash raspiblitz, and try to restore from seed words and old SCB. Is that the recommended way to force close channels you mentioned?

Yes, that is it. From the SCB use the latest (so it has all your channels).
Guide here: https://github.com/rootzoll/raspiblitz/blob/v1.6/FAQ.md#2-recover-from-wallet-seed

bruceleeroy18, 2021-03-24:
I had it setup on drop box. That was a week ago. Will it be latest SCB? I have been unable to get a transfer to USB drive.

Is there a way to get the latest from the HDD?

Sent from ProtonMail mobile

-------- Original Message --------

openoms, 2021-03-24:
the channel.backup only changes when a channel is opened or closed.
it is stored on the disk and copied tp the sdcard too.
See:
`sudo find / -name channel.backup`

bruceleeroy18, 2021-03-24:
Thanks, as soon as I'm home from work I will update you.

Sent from ProtonMail mobile

-------- Original Message --------

bruceleeroy18, 2021-03-25:
Thanks so much. The SCB and seed words restored it.

closed 2021-03-25 by bruceleeroy18