raspiblitz/raspiblitz #2108: Connecting Specter Desktop with the RaspiBlitz

nclapham, 2021-03-25:
Hi, 

I'm connecting Connecting Specter Desktop with the RaspiBlitz, following https://d11n.net/connect-specter-desktor-with-raspiblitz.html.

My setup (Tor off):
RaspiBlitz v1.6.3

Problem: RaspiBlitz doesn't let me connect!
nc -4 -zv 192.168.1.199 8332
 -- nc: connectx to 192.168.1.199 port 8332 (tcp) failed: Connection refused

Even with the following settings (for testing):
 ufw disable (or  ufw allow 8332)

bitcoin.conf
main.rpcport=8332
rpcallowip=0.0.0.0/0
main.rpcbind=0.0.0.0

Could someone give me a hint on where to look next?

openoms, 2021-03-25:
Try this guide: https://github.com/openoms/joininbox/blob/master/prepare_remote_node.md

And see the related issue: https://github.com/rootzoll/raspiblitz/issues/2090

nclapham, 2021-03-25:
Thank you! 
I and new to RaspiBlitz and was editing the incorrect config file. The correct bitcoin.conf is in /mnt/hdd/bitcoin, (not ~/.bitcoin).

Is this the best way to setup Specter Desktop to connect directly from the desktop to the RaspiBlitz bitcoin node?

closed 2021-03-25 by nclapham

openoms, 2021-03-25:
yes, this is the way.

dennisreimann, 2021-03-25:
Thanks for the pointer, corrected it.