raspiblitz/raspiblitz #2117: Freezing on Boot with Pi 4 Fresh Install

ChiefGyk, 2021-03-28:
Attempted to run with the image and it seems to be stuck at 
i2c /dev entries driver and never moves on beyond that. 
For reference it's a Pi 4 4GB and 2GB I tried and I plan to run it without a display when it's all done, but set up is being done with a HDMI monitor.

ChiefGyk, 2021-03-28:
Upon further investigation it looks like this may be similar to #1058

kytmanov, 2021-03-29:
I have similar issue, so when I reboot my node I connect keyboard and press ctrl+d and then it is able to boot.

mantiev, 2021-04-04:
i'm getting the same - fresh install - everything is new - gets stuck at the same point - ctrl+d with keyboard didn't work.  I have used LCD and HDMI and get the same with each.

rootzoll, 2021-04-15:
I am not able to reproduce this problem on my test machines ... but the v1.7RC2 tries to adress this "i2c freeze". @ChiefGyk & @kytmanov can you download the RC2, test and report back? See download here: https://github.com/rootzoll/raspiblitz/blob/dev/README.md#downloading-the-software

nyxnor, 2021-05-03:
No reports on this issue within v1.7.0, pls close this.

closed 2021-05-03 by rootzoll