raspiblitz/raspiblitz #2120: Headless instructions not slowing LCD update

Nerfzooka, 2021-03-30:
[image]
Hey Blitzy Folk!
I assembled and configured my lovely Raspiblitz and it's working AWESOMELY.  I asked it to run tor, Electrum Rust server, and the BTC RPC Explorer with no problems.  
I can SSH into it with nary a glitch so I took off the LCD.  Everything is running smoothly so I decided to venture off the map into RasbiBlitz Commandline.  I looked at a script "00infoLCD.sh" it had comments about headless CPU/log clutter, just add the line "headless=on" into /mnt/hdd/raspiblitz.conf , so I added the line to the file and rebooted.  Everything is still working but the info screen still updates every 20 seconds or so.
[image]

Ben10Brasileiro, 2021-03-31:
Good you are interested in what is running, very nice.
About the update, maintain the default time, there is no problem in that... but if you want to let more time pass for each update, add `sleep 30` (unit is seconds) here https://github.com/rootzoll/raspiblitz/blob/755b7460ca7d8f6aa614748e534ec0941fc891ea/home.admin/00infoBlitz.sh#L306 or here
https://github.com/rootzoll/raspiblitz/blob/755b7460ca7d8f6aa614748e534ec0941fc891ea/home.admin/00mainMenu.sh#L176
I advert this will not improve anything.

I can see that cat looking deep into my eyes.

Nerfzooka, 2021-04-01:
Oh, the main menu script!  I've been looking for that, Thanks Ben10Brasileiro.

rootzoll, 2021-04-06:
Hi there @Nerfzooka thanks for testing on this. The headless mode in v1.6.1 was a quick fix by @PatrickScheich just slowing down the info-loop.

With the now upcoming v1.7 release I refactored the display modes you can run your RaspiBlitz that can be set with:
`sudo /home/admin/config.scripts/blitz.display.sh set-display [lcd|hdmi|headless]`

If you choose `headless` now it will completly deactivate the auto-login of the pi user that had running the infoscreen in loop for HDMI or LCD users and is causing quite some cpu/system load.

This will be part of the v1.7RC2 coming in the next days - happy to get feedback on it.

BTW for people building the sd card themselves the `headless` can now be set already on building the image .. so there is no pi user auto-login and no LCD driver installs from the start.

rootzoll, 2021-04-13:
OK headless mode looks fine on test runs of RC2 - will be part of final v1.7 release.

closed 2021-04-13 by rootzoll