raspiblitz/raspiblitz #2121: 100% cpu LND Log flooding - Got ChannelUpdate for edge not found in graph

14hkr, 2021-03-31:
Hello,  I have noticed high cpu usage on LND for days and log filling with messages below, is it normal or what can be done?
restarting lnd stops messages but after some hours the messages return and 100% cpu
Tia M

2021-03-31 06:39:51.764 [DBG] DISC: Got ChannelUpdate for edge not found in graph(shortChanID=683417944996642817), saving for reprocessing later
2021-03-31 06:39:51.766 [DBG] DISC: Got ChannelUpdate for edge not found in graph(shortChanID=683432238607630336), saving for reprocessing later
2021-03-31 06:39:51.769 [DBG] DISC: Got ChannelUpdate for edge not found in graph(shortChanID=683439935282085889), saving for reprocessing later
2021-03-31 06:39:51.775 [DBG] DISC: Got ChannelUpdate for edge not found in graph(shortChanID=683442134278340609), saving for reprocessing later
2021-03-31 06:39:52.244 [DBG] DISC: Got ChannelUpdate for edge not found in graph(shortChanID=683443233760215041), saving for reprocessing later

Ben10Brasileiro, 2021-03-31:
Check the services running with cpu information with `htop`.

14hkr, 2021-04-01:
It seems to have calmed down, cpu now running at 2-5% for lnd.
Perhaps it needed to clean out bad/old nodes and it took a while.
rpi4 steady at 47-49C and a few % cpu.
Guess i never was a problem. solved

14hkr, 2021-04-01:
Guess i never was a problem. solved

closed 2021-04-01 by 14hkr