raspiblitz/raspiblitz #2122: SSH login not working

Zwoggo, 2021-03-31:
Hello everyone,
I just received my Raspiblitz from Fulmo and am trying to set it up.

Unfortunately, I am not able to connect to the Raspi using SSH on my Windows laptop. I always receive the message:

Connection reset by 192.168.1.104 port 22

I can ping the Raspi. And I think that the issue is not with my router. I have used different SSH clients.
I looked to find information on the issue. However, I am not used to working with a console at all and do not really know what to do here. It seems as if the issue can be with a variety of things. There is lots of info available online. But none of it is worded in a way that a newbie like myself could make any sense of it.

Any recommendations on what to do?

Thanks
Zwoggo

walker377, 2021-03-31:
Hi Zwoggo - Im a complete newbie as well and also just received my Raspiblitz from Fulmo and had the same issue. I resolved it by following this point in the FAQ.
https://github.com/rootzoll/raspiblitz/blob/v1.6/FAQ.md#how-can-i-repair-my-ssh-login

redOz92, 2021-08-25:
Hello there, 

Also very new into this.
So I took out the SD card and it asks me to format it when I put it into my laptop. I can see 2 drives:
1. **boot** --> do I just need to create a text document named ssh.reset and that's it?
2. **secure digital storage** --> can't access this one, it asks me to format it.

Thanks alot.