raspiblitz/raspiblitz #2123: (noHDD) - Connect external HDD/SSD

walker377, 2021-03-31:
Hi Everyone

Ive just got a RaspiBlitz from Fulmo but its not recognising the SSD. Ive run the SSH query and get the following:

If HDD is connected - please report the result of the following command:
sudo /home/admin/config.scripts/blitz.datadrive.sh status
***********************************************************
admin@raspberrypi:~ $ sudo /home/admin/config.scripts/blitz.datadrive.sh status
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=0
isBTRFS=0

# SETUP INFO
isSSD=0
hddCandidate=''
hddBytes=0
hddGigaBytes=0
hddPartitionCandidate=''

# RAID
isRaid=0
raidCandidates=0

# SWAP
isSwapExternal=0

Could this be a power issue? im using the supplied Raspberry PI official USB-C power supply.

Ben10Brasileiro, 2021-03-31:
First try to make sure the cable is plugged sufficiently. Try it again. If still doesnt work, contact Fulmo to report your problem. I use the same SSD with no problems.

walker377, 2021-04-01:
Thanks Ben10Brasileiro

I've checked the SSD is correctly connected and it is so have contacted Fulmo. Unfortunately I don't have another device to test the drive on so dont know if its the drive or the connection on the board that's causing the issue.

nyxnor, 2021-04-18:
@walker377 Was this solved?

walker377, 2021-04-18:
@nyxnor I got the node up and running by buying an external SSD case and connecting via USB. I couldn't get the installed SSD board to work. Waiting for an update from Fulmo on the issue.

nyxnor, 2021-05-03:
@walker377 Pls close this. Hope your problem was solved with the shop by now.

walker377, 2021-05-03:
@nyxnor Will close now as its a hardware issue not a Rasipblitz issue. Still waiting for Fulmo to resolve.

closed 2021-05-03 by walker377