raspiblitz/raspiblitz #2124: RTL installation - error: npm ERR! cb() never called!

pinnubi, 2021-03-31:
Hi,

I am very new to Raspiblitz and just did my first installation on a Raspberry Pi 4B with 8 GB.

There were some initial errors (Stuck at "i2c /dev entries driver" - fixed with '/home/admin/config.scripts/blitz.lcd.sh hdmi on') and I am using another display which I had to install drivers for (waveshare 3.5").

After that, the installation went smoothly and I was able to copy a blockchain from one of my other computers. Raspiblitz is running successfully now (Bitcoin Fullnode + Lightning Network + Tor).

I then tried to install RTL, which did not went through with the following errors:

`dialogcancel(0)
Loop Setting unchanged.
RTL Webinterface Setting changed ..
making sure services are not running
*** INSTALL RTL ***
nodeJS is already installed
Installed nodeJS v14.15.4
adduser: The user `rtl' already exists.
*** make sure rtl is member of lndadmin ***
*** make sure symlink to central app-data directory exists ***
# add the rtl user to the loop group
usermod: group 'loop' does not exist
# symlink the loop.macaroon
*** Get the RTL Source Code ***
Cloning into '/home/rtl/RTL'...
remote: Enumerating objects: 532, done.
remote: Counting objects: 100% (532/532), done.
remote: Compressing objects: 100% (343/343), done.
remote: Total 17138 (delta 345), reused 301 (delta 186), pack-reused 16606
Receiving objects: 100% (17138/17138), 65.90 

MiB | 1.98 MiB/s, done.
Resolving deltas: 100% (14002/14002), done.
HEAD is now at 5a38585b Release 0.10.0 (#571)
OK - RTL code copy looks good

*** Run: npm install ***
[            ......] / extract:rxjs: sill extract rxjs@6.6.3 extracted to /home/rtl/RTL/node_modules/.staging/rxjs-23d43693 (460486ms)`

Here it stops for quite a while, around 5 minutes and then continues with:

`*** Run: npm install ***
npm ERR! cb() never called!

npm ERR! This is an error with npm itself. Please report this error at:
npm ERR!     <https://npm.community>`

As stated, this is of course a npm error, but I thought I report it here.

Cheers and thank you all for your great work!

Ben10Brasileiro, 2021-03-31:
Goon on telegram:

The fast fix didn't do much at first:
```
sudo npm cache clean --force
```
but updating did the trick:
```
npm i -g npm@latest
```
RTL is now installed again.

@openoms Just to mention this npm problem.

Ben10Brasileiro, 2021-04-01:
Should I ping Rootzoll or Openoms on this kind of issues to update the packages?

rootzoll, 2021-04-15:
Can you please test if such fixes are relevant to the v1.7RC2 available for download here: https://github.com/rootzoll/raspiblitz#downloading-the-software

@openoms the nodejs is at the moment 14.15.4 .. should we update 14.16.1 is LTS and current 15.14.0 - see https://nodejs.org/en/

rootzoll, 2021-09-16:
RTL installs looking good now and will be part of 1.7.1rc2. Set to final testing.

rootzoll, 2021-09-21:
The error is back on my tests:

```
3726 silly extract rxjs@7.1.0 extracted to /home/rtl/RTL/node_modules/.staging/rxjs-23d43693 (52589ms)
3727 silly extract @angular/common@12.0.5 extracted to /home/rtl/RTL/node_modules/.staging/@angular/common-f2fc25b3 (56515ms)
3728 silly extract @angular/material@12.0.5 extracted to /home/rtl/RTL/node_modules/.staging/@angular/material-d651ff7c (57294ms)
3729 silly extract typescript@4.2.4 extracted to /home/rtl/RTL/node_modules/.staging/typescript-f5a1a9be (55605ms)
3730 timing npm Completed in 2259211ms
3731 error cb() never called!
3732 error This is an error with npm itself. Please report this error at:
3733 error <https://npm.community>
```
[image]

@openoms do you know what the fix was before?

openoms, 2021-09-22:
when I have seen this it was fixed by updating npm

openoms, 2021-09-22:
fixed above.
```

  npm7installed=$(npm -v 2>/dev/null | grep -c "7.")
  if [ ${npm7installed} -eq 0 ]; then
    # needed for RTL
    # https://github.blog/2021-02-02-npm-7-is-now-generally-available/
    echo "# Update npm to v7"
    sudo npm install --global npm@7
  fi
 ```
 should fix it

rootzoll, 2021-10-09:
OK .. RTL installes working. Closing for release.

closed 2021-10-09 by rootzoll