raspiblitz/raspiblitz #2126: Repair SD card Filesystem Extension

rootzoll, 2021-03-31:
With the change to 64-bit base image of RaspiOS the filesystem extension to the full capacity or the users sd card is not working correctly. To be fixed for final release.

openoms, 2021-04-02:
Here I just extracted the script from raspi-config.
Tested ok with the 64bit pure debian image: https://github.com/openoms/joininbox/blob/master/scripts/standalone/expand.rootfs.sh

rootzoll, 2021-04-03:
The problem was not with the expand itself - the baseimage name changed and so it was not correctly triggered.

Just merged the fix - so this should now be ready for v1.7 release and final testing.

rootzoll, 2021-04-16:
Ok multiple test show that the filesystem is now expanding corretly. Closing the issue.

closed 2021-04-16 by rootzoll